google-deepmind/mujoco #712: Mujoco as vcpkg package.

talregev, 2023-01-26:
I created a vcpkg mujoco port. It mean that other can easily install mujoco library for windows/linux/osx and others triplets.
 There issue need to address for cmake and tinyxml license.  

You can learn more about it in the PR I created and read the discussion, also ask and comment there. 

https://github.com/microsoft/vcpkg/pull/28993/

saran-t, 2023-01-31:
Hi @talregev, thanks for looking into converting MuJoCo into a vcpkg!

Regarding tinyxml2, we do not actually redistribute tinyxml2 source code as part of the MuJoCo repository. The language in the license relating to attribution specifically refers to a source distribution, which isn't what we are making here, and so we feel that adding specific license attribution is not required. (By the way, if you download our [prebuilt binaries](https://github.com/deepmind/mujoco/releases/tag/2.3.1), the THIRD_PARTY_NOTICES file does contain full license attribution for tinyxml2.)

Please feel free to revert to the Microsoft team with this information, hopefully this is sufficient.

talregev, 2023-02-03:
@saran-t 
I see that you remove `TINYXML2_LIB` declaration:
https://github.com/deepmind/mujoco/commit/7251e7b900f0e1b09ce138f9a82ebdf72cc7a6cc
Thank you!

closed 2023-02-03 by talregev