google-deepmind/mujoco #714: When objects are connected, they can hardly be moved.

makitayugo, 2023-01-27:
Hi,

I'm a student and I'm trying to use MuJoCo for master's thesis research.

I'm looking for some help with combination of arm robot with other objects.

Here is a model which explains my question:

<details>
  <summary>minimal XML</summary>

```XML
<mujoco model="panda scene">

  <include file="panda.xml"/>

  <statistic center="0.3 0 0.4" extent="1"/>
  <extension>
    <required plugin="mujoco.elasticity.cable"/>
  </extension>

  <visual>
    <headlight diffuse="0.6 0.6 0.6"  ambient="0.3 0.3 0.3" specular="0 0 0"/>
    <rgba haze="0.15 0.25 0.35 1"/>
    <global azimuth="120" elevation="-30"/>
  </visual>
  <size nconmax="5000" njmax="10000"/>
  
  <option>
      <flag multiccd="disable"/>
  </option>

  <asset>
    <texture type="skybox" builtin="gradient" rgb1="0.3 0.5 0.7" rgb2="0 0 0" width="512"
        height="3072"/>
    <texture type="2d" name="groundplane" builtin="checker" mark="edge" rgb1="0.2 0.3 0.4"
        rgb2="0.1 0.2 0.3" markrgb="0.8 0.8 0.8" width="300" height="300"/>
    <material name="groundplane" texture="groundplane" texuniform="true" texrepeat="5 5"
        reflectance="0.2"/>
  </asset>

  <worldbody>
    <light pos="0 0 1.5" dir="0 0 -1" directional="true"/>
    <geom name="floor" size="0 0 0.05" type="plane" material="groundplane"/>
    <composite type="cable" curve="s" count="41 1 1" size="1.5" offset="-1 0.3 .01"  initial="none">
      <plugin plugin="mujoco.elasticity.cable">
        <!--Units are in Pa (SI)-->
        <config key="twist" value="1e3"/>
        <config key="bend" value="1e4"/>
      </plugin>
      <joint kind="main" damping=".001"/>
      <geom type="capsule" size=".004" rgba=".8 .2 .1 1"  friction=".1" condim="3"/>
    </composite>
    <body name="adapter" pos=".5 0.3 .01">
      <geom type ="sphere" size=".0055 "  friction=".1" condim="3"/>
      <freejoint/>
    </body>
    <body name="box" pos=".5 0 .1">
      <geom type ="box" size=".01 .01 .01 " conaffinity="2" friction=".1" condim="3"/>
      <freejoint/>
    </body>
    <!--<body name="socket" pos="1 0.3 .1">
      <geom type="box" size="0.05 0.05 0.05" margin=".02" gap=".01" rgba=".1 .8 .2 1" contype="2" conaffinity="2"/>
    </body>-->
    <!--<body name="box1" pos="0.1 0.3 .1" zaxis="0 0 1">
      <geom type="box" size="0.05 0.05 0.05"  group="1" condim="3"/>
    </body>
    <body name="box2" pos="0.5 0.3 .1" zaxis="0 0 1">
      <geom type="box" size="0.05 0.05 0.05"  group="1" friction=".1" condim="3"/>
    </body> -->
  </worldbody>
  <equality>
    <connect name="right_boundary" body1="B_last" body2="adapter" anchor=".0376 0 0"/>
  </equality>
  <contact>
    <exclude body1="B_last" body2="adapter"/>
  </contact>>

</mujoco>
```

</details>

Here is a video, illustrating my question:
When the cable is selected and operated, it works as usual, but when the tip sphere (connected by ```epuality```) is selected and operated, the object hardly moves. If the cable is disconnected from the tip sphere and only the sphere is moved, the object can be moved as usual. If the arm robot is not mounted at the same time, the object can be moved as usual.

The first video is a robot and a cable.
The second video is cable only.
The two cables are exactly the same cable.

[grab cable.webm](https://user-images.githubusercontent.com/85473041/215025535-07849e90-2f27-44c8-b0f5-55526cde1efe.webm)
[grab cable３.webm](https://user-images.githubusercontent.com/85473041/215025550-9e8689ad-8403-4207-9a22-ce6981d28962.webm)

quagla, 2023-02-01:
I can reproduce the behaviour from the first video but not of the second one. This behaviour is expected because the perturbation force and damping are proportional to the mass of the object. Try changing the mass of the sphere to get an idea of this effect. You can also change the parameter `stiffness` in the XML (see https://mujoco.readthedocs.io/en/latest/XMLreference.html#visual-map) if you need to apply more force with the mouse via the sphere.

quagla, 2023-02-02:
I'll close the issue but if you think that the behaviour of the cable changes with or without the robots (which I was not able to reproduce), then there is a bug so feel free to reopen it.

closed 2023-02-02 by quagla

makitayugo, 2023-02-07:
I think that the behaviour of the cable changes with or without the robots.
Sorry, we are not able to answer your questions. In my last question I did not put the code with no robot.
This is the program implemented with cables only.
<details>
  <summary>minimal XML</summary>

```XML
<mujoco model="Cable">
  <include file="scene.xml"/>

  <extension>
    <required plugin="mujoco.elasticity.cable"/>
  </extension>

  <statistic center="0 0 .3" extent="1"/>
  <visual>
    <global elevation="-30"/>
  </visual>

  <compiler autolimits="true"/>

  <size memory="2M"/>

  <worldbody>
    <!--Cable-->
    <composite type="cable" curve="s" count="41 1 1" size="1.5" offset="-1 0.3 .01"  initial="none">
      <plugin plugin="mujoco.elasticity.cable">
        <!--Units are in Pa (SI)-->
        <config key="twist" value="1e3"/>
        <config key="bend" value="1e4"/>
      </plugin>
      <joint kind="main" damping=".001"/>
      <geom type="capsule" size=".004" rgba=".8 .2 .1 1"  friction=".1" condim="3"/>
    </composite>
    <body name="adapter" pos=".5 0.3 .01">
      <geom type ="sphere" size=".0055 " conaffinity="2" friction=".1" condim="3"/>
      <freejoint/>
    </body>

    <!--Adhesive socket-->
    <!--body name="socket" pos=".7 0 .5">
      <geom type="box" size=".02 .02 .02" margin=".01" gap=".01" rgba=".1 .8 .2 1" contype="2" conaffinity="2"/>
    </body>-->

    <!--Non-adhesive box-->
    <body name="nonadhesive" pos="1.2 0 .1">
      <geom size=".01" rgba=".1 .2 .8 1"/>
      <freejoint/>
    </body>
  </worldbody>

  <!--Attaching adapter to the end of cable-->
  <equality>
    <connect name="right_boundary" body1="B_last" body2="adapter" anchor=".0376 0 0"/>
  </equality>

  <contact>
    <exclude body1="B_last" body2="adapter"/>
  </contact>

  <!--Making socket adhesive-->
  <!--<actuator>
    <adhesion name="adhere_socket" body="socket" ctrlrange="0 1" gain="1"/>
  </actuator>-->
  
</mujoco>
```