google-deepmind/mujoco #717: Modeling a floor with different friction at different locations

aravindsiv, 2023-01-29:
Hi,

I'm a student trying to use MuJoCo to model a vehicle on uneven surfaces.

I'm looking for help loading a floor where some parts may have a different friction coefficient than others. I can define the friction coefficient given the surface normal's (x,y) coordinate.

Do you have any pointers on how to go about this?

erez-tom, 2023-01-29:
This feature isn't supported, but I can suggest how you may achieve it:
The constraint resolution solver reads the [friction coefficient](https://github.com/deepmind/mujoco/blob/main/include/mujoco/mjdata.h#L74) from the [contact array](https://github.com/deepmind/mujoco/blob/main/include/mujoco/mjdata.h#L312).  The value in the contact array is resolved from the friction coefficients of the two colliding geoms (e.g., [here](https://github.com/deepmind/mujoco/blob/main/src/engine/engine_collision_driver.c#L669) and elsewhere in those code blocks).  At any stage (e.g., [here](https://github.com/deepmind/mujoco/blob/main/src/engine/engine_forward.c#L802)) after collision detection (which is part of `mj_fwdPosition`) and before constraint resolution (which is part of `mj_fwdConstraint`) you can mutate the contact array (every entry tells you the contact's [position](https://github.com/deepmind/mujoco/blob/main/include/mujoco/mjdata.h#L69), so you can use that to edit the friction value).

If this proves useful, we can consider adding a "collision" plug-in entry point, for users to conveniently interject the computation pipeline and edit the contact array.

closed 2023-01-29 by erez-tom

aravindsiv, 2023-01-29:
Thanks! Adding the collision plug-in entry point would be helpful, but I will check out your suggestion.

yuvaltassa, 2023-01-29:
What about making the floor out of boxes with different friction / contact properties?

aravindsiv, 2023-01-29:
That was my first go-to, but it creates a discontinuity in friction as the robot moves from one box to the next.

For my particular use case, defining a rolling friction coefficient as a smooth function over the (x,y) of the gem's surface normal would be preferable.