google-deepmind/mujoco #718: Using ```print(physics.named.data.geom_xpos)```, we could get the coordinates of the object, but the coordinates of the initial point will always be displayed, even if the object is manipulated and its position changes. Is there any way to display the coordinates accordingly when the object is moved?

makitayugo, 2023-01-30:
Using ```print(physics.named.data.geom_xpos)```, we could get the coordinates of the object, but the coordinates of the initial point will always be displayed, even if the object is manipulated and its position changes. Is there any way to display the coordinates accordingly when the object is moved?

_Originally posted by @makitayugo in https://github.com/deepmind/mujoco/issues/678#issuecomment-1408152649_

quagla, 2023-02-07:
Do you have a minimal example to reproduce this issue?

makitayugo, 2023-02-08:
・Using line 53, the coordinates are updated each time the object moves, but I do not know how to name the coordinates.
・Using line 54, I can name the coordinates, but the coordinates are not updated when the object moves.

Is there a way to update the coordinates with the coordinates named?

<details>
  <summary>minimal XML</summary>

```XML
import argparse

import mujoco
import mujoco_viewer

from dm_control import mujoco

# Access to enums and MuJoCo library functions.
from dm_control.mujoco.wrapper.mjbindings import enums
from dm_control.mujoco.wrapper.mjbindings import mjlib

# Composer high level imports
from dm_control import composer
from dm_control.composer.observation import observable
from dm_control.composer import variation

# Imports for Composer tutorial example
from dm_control.composer.variation import distributions
from dm_control.composer.variation import noises
from dm_control.locomotion.arenas import floors

# Control Suite
from dm_control import suite

# Run through corridor example
from dm_control.locomotion.walkers import cmu_humanoid
from dm_control.locomotion.arenas import corridors as corridor_arenas
from dm_control.locomotion.tasks import corridors as corridor_tasks

# Soccer
from dm_control.locomotion import soccer

# Manipulation
from dm_control import manipulation

def main(path: str = 'humanoid.xml'):
    model = mujoco.MjModel.from_xml_path(path)
    data = mujoco.MjData(model)
    physics = mujoco.Physics.from_xml_path(path)
    

    # create the viewer object
    viewer = mujoco_viewer.MujocoViewer(model, data)
    

    # simulate and render
    for _ in range(10000):
        if viewer.is_alive:
            mujoco.mj_step(model, data)
            #mujoco.mj_step(physics)
            viewer.render()
            #print(data.geom_xpos)
            print(physics.named.data.xpos)
        else:import argparse

    # close
    viewer.close()

if __name__ == "__main__":
    parser = argparse.ArgumentParser(description='Simulate')
    parser.add_argument('--path', type=str, help='Relative path to an XML file')
    args = parser.parse_args()

    if args.path is None:
        main()
    else:
        main(args.path)
```

quagla, 2023-02-08:
I would suggest to either use `mujoco` or `dm_control` but not mix the two. It looks like you are stepping with `mujoco` and then ask `dm_control` to read the data.

closed 2023-02-09 by quagla

makitayugo, 2023-02-10:
I understand. I would like to use ``mujoco`` (without physics) to display the coordinates of an object.
As I mentioned in my last question, using line 53, the coordinates are updated each time the object moves, but I do not know how to name the coordinates.
<details>
  <summary>minimal XML</summary>

```XML
import argparse

import mujoco
import mujoco_viewer
def main(path: str = 'humanoid.xml'):
    model = mujoco.MjModel.from_xml_path(path)
    data = mujoco.MjData(model)
#physics = mujoco.Physics.from_xml_path(path)
    

    # create the viewer object
    viewer = mujoco_viewer.MujocoViewer(model, data)
    

    # simulate and render
    for _ in range(10000):
        if viewer.is_alive:
            mujoco.mj_step(model, data)
            #mujoco.mj_step(physics)
            viewer.render()
            print(data.geom_xpos)
            #print(physics.named.data.xpos)
        else:import argparse

    # close
    viewer.close()

if __name__ == "__main__":
    parser = argparse.ArgumentParser(description='Simulate')
    parser.add_argument('--path', type=str, help='Relative path to an XML file')
    args = parser.parse_args()

    if args.path is None:
        main()
    else:
        main(args.path)
```
The execution screen in the above program looks like the picture below. We want to name each coordinate so that we know which object it represents.