google-deepmind/mujoco #719: Comparing sensors data between Python and Simulate

itamar-dw, 2023-01-31:
Hi,
I am comparing the numerical values of accelerometer and gyro sensors with the visualization in the Simulate app, and get different results. I want to understand where does this difference come from (my code or something else). I'll focus on gyro since it's enough to demonstrate the issue.

Here is a model which explains my question:

<details>
  <summary>Mujoco  XML model</summary>

```XML
<mujoco model="sensors_test_no_gravity">
  <option gravity="0 0 0"/>
  <asset>
    <texture name="grid" type="2d" builtin="checker" rgb1=".1 .2 .3" rgb2=".2 .3 .4" width="300" height="300"/>
    <material name="grid" texture="grid" texrepeat="8 8" reflectance=".2"/>
  </asset>

  <worldbody>
    <geom name="floor" size="2 2 .01" type="plane" material="grid"/>
    <light name="top" pos="0 0 2"/>

    <body name="box_and_sphere"  pos="0 0 1" euler="0 0 0">
      <site name="imu1" pos="0 0 .2" rgba="0. 1. 0. .5"/>
      <joint name="ball_joint" type="ball" pos="0 0 0" />
      <geom name="red_box" type="box" size=".1 .1 .2" rgba="1 0 0 1"/>
    </body>
  </worldbody>
  <sensor>
      <gyro name="gyro" site="imu1" />
  </sensor>
  <keyframe>
    <key name="init_qvel" qvel="1 1 1" />
  </keyframe>
</mujoco>
```
</details>
In Simulate app I load the keyframe, then start the simulation for about 4 sec.
In Python, I load the keyframe in the code, and run for 4 sec:
<details>
  <summary>Python code</summary>

```Python
import mujoco
import numpy as np

with open('test.xml', 'r') as f:
    xml = f.read()
model = mujoco.MjModel.from_xml_string(xml)
data = mujoco.MjData(model)
mujoco.mj_resetDataKeyframe(model, data, 0) 

gyro = []
ts = []
while data.time < 4:
    mujoco.mj_step(model, data)
    gyro.append(data.sensor(0).data.copy())
    ts.append(data.time)
  
gyro = np.vstack(gyro)
ts = np.r_[ts]
```
</details>

When I plot `gyro` I see that:
1. The `z` component is constant
2. The `x` component grows above the value of 1 and then drops to 0
[image]

However, when looking at the sensor visualtization in Simulate, I see that
1. The `x` component does not grow above 1
2. The `z` component oscillates between 1 and ~0.7. 

So, the `x` behavior may be explained if the value is just clipped to 1 by the graph's y-axis upper limit (that is, a visualization issue). But the behavior of `z` component cannot be explained. (btw as a sanity check I also compared with `data.qvel` and not surprisingly got the same values as in `data.sensor(0).data`).

Sensor visualization at `t=1.3`, demonstrating both issues:
[image]

yuvaltassa, 2023-02-01:
Interesting! This could either be due to a bug in simulate or in the Python bindings.

Could you please do the following two checks:

1. In `simulate` please use the "print data" button to save the entire state of `mjData` to a text file.
2. In Python, instead of using the accessor `data.sensor(0).data`, please use `data.sensordata`.

The numbers that you see correspond to which values?

yuvaltassa, 2023-02-01:
I took a look a quick look myself.
This is disturbing. Marking as a bug.

itamar-dw, 2023-02-01:
> Interesting! This could either be due to a bug in simulate or in the Python bindings.
>
> Could you please do the following two checks:
>
>     1. In `simulate` please use the "print data" button to save [...]

I guess you already saw it: Only the visualization does not match, all other numbers do match

yuvaltassa, 2023-02-01:
But does it make sense that the z-axis value is constant?
Intuitively I would expect them all to oscillate, even a little bit, right?

itamar-dw, 2023-02-01:
> But does it make sense that the z-axis value is constant? Intuitively I would expect them all to oscillate, even a little bit, right?

Yes, in torque-free rotation (which is the case since I set `gravity="0 0 0"`) the `z` component of the angular velocity is a constant of motion. See for example https://phys.libretexts.org/Bookshelves/Classical_Mechanics/Variational_Principles_in_Classical_Mechanics_(Cline)/13%3A_Rigid-body_Rotation/13.20%3A_Torque-free_rotation_of_an_inertially-symmetric_rigid_rotor

itamar-dw, 2023-02-01:
> > But does it make sense that the z-axis value is constant? Intuitively I would expect them all to oscillate, even a little bit, right?
> 
> Yes, it makes sense since in torque-free rotation (which is the case since I set `gravity="0 0 0"`) the `z` component of the angular velocity is a constant of motion. See for example https://phys.libretexts.org/Bookshelves/Classical_Mechanics/Variational_Principles_in_Classical_Mechanics_(Cline)/13%3A_Rigid-body_Rotation/13.20%3A_Torque-free_rotation_of_an_inertially-symmetric_rigid_rotor
> 
> (Actually it should be the case also with gravity, since gravity operates on the center-of-mass so it is still torque-free)

Just to be clear, this holds because of the inertial symmetry of the specific body I use (box with equal width and depth)

closed 2023-02-03 by copybara-service[bot]

yuvaltassa, 2023-02-03:
The was: the plot was _not_ being clipped, but the y-axis didn't have enough precision to show that. I.e., the y-axis was changing but you couldn't see it. So this is now fixed.

commit bc0184d6 (2023-02-03): Increase precision of `simulate` sensor plot y-axis by 1 digit.

- Before this change, the y-axis scale had only one digit, leading to user confusion. Small changes in senor values could appear as if the values remained at the nearest integer value, while in fact the y-axis itself was changing, but this was not evident in the GUI due to the reduced precision.
- Fixes #719.

PiperOrigin-RevId: 506841881
Change-Id: I583036bba389cef8d9ecf1c1bbcfb573ef2285e6