google-deepmind/mujoco #720: FatalError: Rank-defficient Hessian in HessianDirect

shhra, 2023-02-02:
Hi,

I'm a student and I'm trying to use MuJoCo for performing inverse dynamics in Motion Capture data for human skeletal data.

I'm looking for some help building the parser and visualizing the motion capture data. I have built a parser that loads a small ".bvh" file. However, when I want to use my actual data (full body motion capture), I get following error: `mujoco.FatalError: Rank-defficient Hessian in HessianDirect`

Here is a model which explains my question:

```XML

<mujoco model="Mocap data">

<visual>
<rgba haze=".3 .3 .3 1"/>
</visual>

<default>
<joint type="hinge" pos="0 0 0" axis="0 0 1" limited="true" range="0 120" damping="0.1"/>
<muscle ctrllimited="true" ctrlrange="0 1"/>
</default>

<asset>
<texture type="skybox" builtin="gradient" rgb1="0.3 0.6 0.9" rgb2="0 0 0" width="512" height="512"/>
<texture name="texplane" type="2d" builtin="checker" rgb1=".85 .85 .85" rgb2=".3 .3 .3" width="512" height="512" mark="cross" markrgb=".8 .8 .8"/>
<material name="matplane" reflectance="0.3" texture="texplane" texrepeat="1 1" texuniform="true"/>
</asset>

<worldbody>
<geom name="floor" pos="0 0 -0.5" size="0 0 1" type="plane" material="matplane"/>

<light directional="true" diffuse=".8 .8 .8" specular=".2 .2 .2" pos="0 0 5" dir="0 0 -1"/>

<site name="s0" pos="-0.15 0 0" size="0.02"/>
<site name="x0" pos="0 -0.15 0" size="0.02" rgba="0 .7 0 1" group="1"/>

<body pos="0 0 0">

<joint name="Bone" type="hinge" pos="0 0 0"/>
<geom name="Bone" type="cylinder" pos="0 0 0" size="0.02 0.02" mass="0.1"/>
<body pos="0 0 0">

        <joint name="Bone_Pelvis" type="hinge" pos="0 0 0"/>
        <geom name="Bone_Pelvis" type="cylinder" pos="0 0 0" size="0.02 0.02" mass="0.1"/>
        <body pos="-2.89223 81.9919 2.34413">
        <geom name="Bone-Bone_Pelvis" type="capsule" size="0.01" fromto="0 0 0 0.5 0.5 0.5"/>

                    <joint name="Bone_Spine" type="hinge" pos="0 0 0"/>
                    <geom name="Bone_Spine" type="cylinder" pos="0 0 0" size="0.02 0.02" mass="0.1"/>
                    <body pos="0.0 0.0 0.0">
                    <geom name="Bone_Pelvis-Bone_Spine" type="capsule" size="0.01" fromto="0 0 0 11.9606 -0.0119152 1.62455e-05"/>

                        <joint name="Bone_Spine1" type="hinge" pos="0 0 0"/>
                        <geom name="Bone_Spine1" type="cylinder" pos="0 0 0" size="0.02 0.02" mass="0.1"/>
                        <body pos="11.9606 -0.0119152 1.62455e-05">
                        <geom name="Bone_Spine-Bone_Spine1" type="capsule" size="0.01" fromto="0 0 0 14.9703 -0.0119197 0.0"/>

                            <joint name="Bone_Spine2" type="hinge" pos="0 0 0"/>
                            <geom name="Bone_Spine2" type="cylinder" pos="0 0 0" size="0.02 0.02" mass="0.1"/>
                            <body pos="14.9703 -0.0119197 0.0">
                            <geom name="Bone_Spine1-Bone_Spine2" type="capsule" size="0.01" fromto="0 0 0 14.9703 -0.0119197 0.0"/>

                                        <joint name="Bone_Neck" type="hinge" pos="0 0 0"/>
                                        <geom name="Bone_Neck" type="cylinder" pos="0 0 0" size="0.02 0.02" mass="0.1"/>
                                        <body pos="14.9703 -0.0119197 0.0">
                                        <geom name="Bone_Spine2-Bone_Neck" type="capsule" size="0.01" fromto="0 0 0 15.0383 -0.00472953 0.0"/>

                                            <joint name="Bone_Head" type="hinge" pos="0 0 0"/>
                                            <geom name="Bone_Head" type="cylinder" pos="0 0 0" size="0.02 0.02" mass="0.1"/>
                                            <body pos="15.0383 -0.00472953 0.0">
                                            <geom name="Bone_Neck-Bone_Head" type="capsule" size="0.01" fromto="0 0 0 5.87259 0.0 0.0"/>
[... 473 lines cut ...]
                <geom name="Spine-RightArm" type="capsule" size="0.01" fromto="0 0 0 -0.582886 0.347082 0.261168"/>

                    <joint name="RightHand" type="hinge" pos="0 0 0"/>
                    <geom name="RightHand" type="cylinder" pos="0 0 0" size="0.02 0.02" mass="0.1"/>
                    <body pos="-0.582886 0.347082 0.261168">
                    <geom name="RightArm-RightHand" type="capsule" size="0.01" fromto="0 0 0 -0.178315 -0.6159 0.180521"/>

                        <joint name="EndSite_4" type="hinge" pos="0 0 0"/>
                        <geom name="EndSite_4" type="cylinder" pos="0 0 0" size="0.02 0.02" mass="0.1"/>
                        <body pos="-0.178315 -0.6159 0.180521">
                        <geom name="RightHand-EndSite_4" type="capsule" size="0.01" fromto="0 0 0 1.0 0.0 0.0"/>
                    </body>
                </body>
            </body>
        </body>
    </body>
</body>

</body>

</worldbody>
</mujoco>
```

Note: I haven't loaded the animation frames yet. I just wanted to verify if my character skeleton gets successfully built.

saran-t, 2023-02-02:
At first glance, I suspect that this is caused by the `Neck` and `Spine` joints having identical location and axis.

shhra, 2023-02-02:
Thank you! I think it must be the case. :+1:  May I ask you why you suspected it was the case? I am relatively new to the physics simulation. Therefore, I do not have clue about it. I skipped the `Spine2` and it seemed to work. :v:

saran-t, 2023-02-02:
If you're familiar with numerical optimisation, the short answer is that aliasing joints introduce a zero mode. There are continuously many equivalent ways to split the same rotation across the two joints.

shhra, 2023-02-02:
Thank you for clarifying it.

shhra, 2023-02-02:
I found that there was issue with how I ordered the site and joints. I fixed that along with some other joint types for the bones. Consequently I was able to import the skeleton. Thank you for pointing in the right direction. 

[image]

closed 2023-02-03 by saran-t