google-deepmind/mujoco #721: A rendering bug

emailweixu, 2023-02-03:
In the model, the rendering of the box is affected by the ground plane. Without the ground plane, it is correctly rendered. With the ground plane, part of the box become transparent. What I expect is that box should not become transparent.

The model and rendering code: https://colab.research.google.com/drive/1yXi683FxQxgWAf8U9eH52BqaW13roBAK

Rendered without ground plane
[image]
Rendered image without ground plane
[image]

- Include the following context:
  - Operating system: Ubuntu 20.04.5
  - MuJoCo version: 2.3.1
  - For Python issues, what bindings are you using: mujoco

yuvaltassa, 2023-02-03:
This is due to the far and near clipping planes that are affected by `extent` which is automatically computed unless specified by the user.

Search the documentation for "znear" "zfar" and "extent" and play with your model in the interactive viewer while moving the camera around.

Some general background in how openGL does rendering would also help (read about the frustum).

closed 2023-02-03 by yuvaltassa

emailweixu, 2023-02-03:
Thanks for the prompt reply. Setting znear to a smaller value does solve the problem.

> This is due to the far and near clipping planes that are affected by `extent` which is automatically computed unless specified by the user.
>
> Search the documentation for "znear" "zfar" and "extent" [...]