google-deepmind/mujoco #724: I think that the behaviour of the cable changes with or without the robots.

makitayugo, 2023-02-07:
I think that the behaviour of the cable changes with or without the robots.
Sorry, we are not able to answer your questions. In my last question I did not put the code with no robot.
This is the program implemented with cables only.
<details>
  <summary>minimal XML</summary>

```XML
<mujoco model="Cable">
  <include file="scene.xml"/>

  <extension>
    <required plugin="mujoco.elasticity.cable"/>
  </extension>

  <statistic center="0 0 .3" extent="1"/>
  <visual>
    <global elevation="-30"/>
  </visual>

  <compiler autolimits="true"/>

  <size memory="2M"/>

  <worldbody>
    <!--Cable-->
    <composite type="cable" curve="s" count="41 1 1" size="1.5" offset="-1 0.3 .01"  initial="none">
      <plugin plugin="mujoco.elasticity.cable">
        <!--Units are in Pa (SI)-->
        <config key="twist" value="1e3"/>
        <config key="bend" value="1e4"/>
      </plugin>
      <joint kind="main" damping=".001"/>
      <geom type="capsule" size=".004" rgba=".8 .2 .1 1"  friction=".1" condim="3"/>
    </composite>
    <body name="adapter" pos=".5 0.3 .01">
      <geom type ="sphere" size=".0055 " conaffinity="2" friction=".1" condim="3"/>
      <freejoint/>
    </body>

    <!--Adhesive socket-->
    <!--body name="socket" pos=".7 0 .5">
      <geom type="box" size=".02 .02 .02" margin=".01" gap=".01" rgba=".1 .8 .2 1" contype="2" conaffinity="2"/>
    </body>-->

    <!--Non-adhesive box-->
    <body name="nonadhesive" pos="1.2 0 .1">
      <geom size=".01" rgba=".1 .2 .8 1"/>
      <freejoint/>
    </body>
  </worldbody>

  <!--Attaching adapter to the end of cable-->
  <equality>
    <connect name="right_boundary" body1="B_last" body2="adapter" anchor=".0376 0 0"/>
  </equality>

  <contact>
    <exclude body1="B_last" body2="adapter"/>
  </contact>

  <!--Making socket adhesive-->
  <!--<actuator>
    <adhesion name="adhere_socket" body="socket" ctrlrange="0 1" gain="1"/>
  </actuator>-->
  
</mujoco>
```

_Originally posted by @makitayugo in https://github.com/deepmind/mujoco/issues/714#issuecomment-1420298896_

quagla, 2023-02-07:
What is scene.xml?

makitayugo, 2023-02-08:
I'm sorry. ```scene.xml``` is this XML file.
<details>
  <summary>minimal XML</summary>

```XML
<mujoco>
  <compiler meshdir="asset" texturedir="asset"/>
  <statistic extent="1.5" meansize=".05"/>

  <option timestep="0.005" jacobian="sparse"/>

  <visual>
    <rgba haze="0.15 0.25 0.35 1"/>
    <quality shadowsize="4096"/>
    <map stiffness="700" shadowscale="0.5" fogstart="10" fogend="15" zfar="40" haze="0.3"/>
  </visual>

  <asset>
    <texture type="skybox" builtin="gradient" rgb1="0.3 0.5 0.7" rgb2="0 0 0" width="512" height="512"/>
    <texture name="texplane" type="2d" builtin="checker" rgb1=".2 .3 .4" rgb2=".1 0.15 0.2"
      width="512" height="512" mark="cross" markrgb=".8 .8 .8"/>
    <texture name="texcarpet" type="2d" file="carpet.png"/>
    <texture name="texsponge" type="2d" file="sponge.png"/>
    <texture name="texmarble" type="cube" file="marble.png"/>

    <material name="matplane" reflectance="0.3" texture="texplane" texrepeat="1 1" texuniform="true"/>
    <material name="matcarpet" texture="texcarpet"/>
    <material name="matsponge" texture="texsponge" specular="0.3"/>
    <material name="matmarble" texture="texmarble" rgba=".7 .7 .7 1"/>
  </asset>

  <worldbody>
    <light directional="true" diffuse=".4 .4 .4" specular="0.1 0.1 0.1" pos="0 0 5.0" dir="0 0 -1" castshadow="false"/>
    <light directional="true" diffuse=".6 .6 .6" specular="0.2 0.2 0.2" pos="0 0 4" dir="0 0 -1"/>

    <geom name="ground" type="plane" size="0 0 1" pos="0 0 0" quat="1 0 0 0" material="matplane" condim="1"/>

    <body mocap="true" pos="0 0 0" zaxis=".5 0 1">
      <geom type="capsule" size=".1 .1" material="matmarble" group="1" condim="1"/>
    </body>
  </worldbody>
</mujoco>
```

quagla, 2023-02-08:
Thank you. Are you sure that you are applying the same perturbation? The initial viewpoint is very different in the two examples (with and without the arm) you provided so it's easy to draw a very different arrow with your mouse and think that the behavior is different. Could you please double check that you are applying the same perturbation (i.e. the actual size of the arrows is comparable)?

closed 2023-02-09 by quagla