google-deepmind/mujoco #727: Discrete elastic rods, mass-spring model, and fluid simulation

JohnChen1024a, 2023-02-08:
1. What model does the elasticity plugin use? Any reference?
2. Will MuJoCo implement the "discrete elastic rods"?
3. Will MuJoCo implement fluid simulation? How about fluid interaction with join objects, such as swimming under water?
4. A simple example for mass-spring simulation of rods.

quagla, 2023-02-08:
Hi, apologies for the limited plugin documentation, we are planning to improve it in the future once the shell plugin is also released. Concerning your questions:

1.  The 3D elasticity is documented in the comments e.g. https://github.com/deepmind/mujoco/blob/main/plugin/elasticity/solid.cc#L78 it is a https://en.wikipedia.org/wiki/Hyperelastic_material#Saint_Venant%E2%80%93Kirchhoff_model formulated in a coordinate-free manner. See Weischedel or Quaglino (myself) PhD theses for more details of the 2D case.
2. I am not planning to at the moment. We are using a reduced coordinate formulation based on quaternions that results in simpler gradients and a true inextensible formulation, allowing large time steps with implicit damping.
3. There is a simple fluid model but it's not documented. See `fluidshape` and `fluidcoef` in https://mujoco.readthedocs.io/en/latest/XMLreference.html.
4. Do you need the rods to be extensible? Do you need them to form a close loop?

JohnChen1024a, 2023-02-09:
Thank you for your very useful information. I will check that. 
3. For the fluidshape and fluidcoef, it seems that they provide a fluid interaction model to objects, but not for fluid simulation. If it is true, hope MuJoCo will implement fluid simulation. Otherwise, would you provide a simple example, such as dropping a cube into a water container?
4. For extensible or non-extensible rods, provide a simple example. I do not need them to form a close loop.

quagla, 2023-02-09:
A fluid simulator is not high on our priorities at the moment unfortunately. I don't think it will happen in the first half of this year.

If you look in https://github.com/deepmind/mujoco/tree/main/model/plugin, the examples `belt`, `cable`, and `coil` are very similar to those found in the Discrete Elastic Rod paper. In fact, you should be able to get the same results as in that paper since the underlying continuous model is the same. We are using a different set of coordinates that does not require to write the stretching energy because inextensibility is enforced by construction.

yuvaltassa, 2023-02-09:
Regarding examples of the current stateless fluid models (there are two, actually) see [here](https://github.com/deepmind/mujoco/blob/main/test/engine/testdata/derivative/tumbling_thin_object.xml) and [here](https://github.com/deepmind/mujoco/blob/main/test/engine/testdata/derivative/tumbling_thin_object_ellipsoid.xml)

JohnChen1024a, 2023-02-09:
@quagla Does it mean that this model is faster than that of Discrete Elastic Rod? What is the reason this model cannot simulate closed loops? The simulations of belt, cable and coil look good. Thanks.

JohnChen1024a, 2023-02-09:
@yuvaltassa Thank you for the examples.

quagla, 2023-02-14:
We basically have the same number of degrees of freedom, which is O(4V): for Discrete Elastic Rods it's coming from 3 vertex positions + 1 twist per edge, while for us it's one quaternion (4 components) per edge + the position of only the first vertex, so I would expect them to have the same performance. 

One advantage of our model is the possibility to add diagonal implicit damping, improving the stability. The disadvantage is that we cannot do closed loops because each quaternion, which specifies the position of each segment with respect to the previous one, can have any value if the chain is open. If closed (which is an additional constraint), then the each quaternion influences all other ones, including its previous one and itself. There is no known way to write the resulting reduced configuration space in an analytic form.

JohnChen1024a, 2023-02-16:
@quagla Thank you for your very good explanation. Wish you overcome the problem of closed loops soon. 

4. A simple example for mass-spring simulation of rods. For extensible or non-extensible rods, provide a simple example. I do not need them to form a close loop. I would like to know in the reduced coordinate formulation based on quaternions, how to formulate these simple examples, and simulate them in MuJoCo, similar to the following example.

https://user-images.githubusercontent.com/124339023/219418987-22edb683-6bd4-4a8e-894b-e43fcdfe633b.mp4

quagla, 2023-02-19:
At the moment we don't support stretching of the cable, since this is usually negligible for bending deformations (basically all deformations where the cable is not pulled). Conversely, if you are in the pulled-cable scenario, I would recommend using a tendon and neglect the bending contribution.

It is however possible if desired to add a slider dof to simulate stretching in the cable, but this involve modifying https://github.com/deepmind/mujoco/blob/main/plugin/elasticity/cable.cc Do you feel like giving it a go?

JohnChen1024a, 2023-02-23:
1. For the above simple example, there are ten particles with mass which are connected with springs, following Hooke’s law. The first particle moves horizontally. How to formulate and simulate this example in MuJoCo?
2. For my simulation of rods or cables, actually, it is preferred no stretching.
3. For your cable model, is it good to simulate human hairs or a brush?
4. Adding a slider Dof to simulate stretching in the cable, it is a good extension. However, I am still a beginner for MuJoCo.

quagla, 2023-03-01:
1. This can be done with tendons
2. The I would highly recommend to use the cable
3. Absolutely yes
4. It's not needed given what you say on point 2 and your objects of interest on point 3

Looking forward to your simulations!

JohnChen1024a, 2023-03-03:
Thanks for all your helpful answers. I will look at detail of cables and tendons.

quagla, 2023-03-03:
I will close this for now, feel free to re-open if you have further questions.

closed 2023-03-03 by quagla

haasithp, 2023-06-01:
> At the moment we don't support stretching of the cable, since this is usually negligible for bending deformations (basically all deformations where the cable is not pulled). Conversely, if you are [...]

Could you please give some insights on simulating stretching in the cable?

giorgionicola, 2024-04-09:
Hi, is there any update about stretching cables?