google-deepmind/mujoco #728: Time derivate with respect to a period of time using `mjd_transitionFD`

rongakowang, 2023-02-09:
Hi,

I am a student and am looking for help related to the time derivate of the states for a period of time. My understanding is that `mjd_transitionFD` can calculate the derivate of `x_{t+h}` with respect to `x_t` where `h` is a one-time step (i.e. by default 0.002). However, what should I do if I wish to get the derivate for a continuous period of time `T` (say `T = 1s`, which contains 500 simulation steps then), namely the derivate of `x_{t+T}` with respect to the initial state `x_0`? If I were to multiply all these (500) one-step derivates matrices, I would end up with an inf total derivate: I thus wonder what's the correct way to do so

```python
total_derivative = np.eye(2*model.nv)
A = np.zeros((2*model.nv, 2*model.nv))
while data.time < DURATION:
  # Step the simulation.
  mujoco.mj_step(model, data)
  mujoco.mjd_transitionFD(model, data, 1e-6, True, A, None, None, None)

  total_derivative = A @ total_derivative # 
# np.linalg.norm(total_derivative) = inf
```

Also, I wonder what's the correct way in MuJoco to define a "fixed" articulated body, which will remain at 0 velocities and at any time even with the existence of contact and gravity, but you can still set the `qpos` to control each of its child links? (like a firm hand floating in the air and won't move when another object `A` hit it?). My current attempt is to set `gravcomp` for the body and specify a tiny range in `joint`:

```xml
<body name="link6" pos="0 0 0" quat="0.765218 0 0 -0.643772" gravcomp="1">
       <inertial pos="0 0 0" mass="0.2001" diaginertia="2 2 2"/>
       <joint name="joint6" pos="0 0 0" axis="0 0 1" range="-0.0001 0.0001"/>
       <geom type="mesh" mesh="mesh1"/>
       <body name="link8" pos="-0.0880972 -0.00520036 0.020686" quat="0.999976 0.00686852 0 0" gravcomp="1">
```

If I am to introduce such an object, will it cause issues to the derivative of `A`?

Thank you in advance for any hints on this issue

rongakowang, 2023-02-10:
For the first question, I notice that `mj_step` can be made to call with `nstep`, will `mjd_transitionFD` support similar arguments?

yuvaltassa, 2023-02-10:
1. Your approach to multiplying the derivatives is correct.
2. Try to give you joint a very high armature.
3. No, derivatives over multiple timesteps will not be supported. Multiplying the A matrices is correct.

closed 2023-02-10 by yuvaltassa

EladSharony, 2023-11-27:
For those seeking to calculate a derivative over multiple time steps (`x_{t+N*h}`), here's a hacky workaround:

1. Initialize `x_{t}` and `u_{t}` in your mjData.
2. Create two arrays, `A` and `B`, using: `A, B = np.zeros((model.nq + model.nv, model.nq + model.nv)), np.zeros((model.nq + model.nv, model.nu))`.
3. Increase the simulation time step by a factor of `N`: `model.opt.timestep *= N`.
4. Execute `mj.mjd_transitionFD(model, data, 1e-12, 1, A, B, None, None)` for the transition.
5. Revert the simulation time step to its original value: `model.opt.timestep /= N`.

**However, keep in mind:**
- **Using a large N*h value might lead to inaccurate results.**
- **It's advisable to verify the results by manually calculating `A` and `B`.**

In my case, the following worked fine:
- `Integrator`: `"implicitfast"`.
- `Time step`: `0.0025`.
- `N` value: `20` (`N*h=20*0.025=0.05`).