google-deepmind/mujoco #731: Display of coordinates according to object movement.

makitayugo, 2023-02-10:
I understand. I would like to use ``mujoco`` (without physics) to display the coordinates of an object.
As I mentioned in my last question, using line 53, the coordinates are updated each time the object moves, but I do not know how to name the coordinates.

The execution screen in the below program looks like the picture below. We want to name each coordinate so that we know which object it represents.

[image]

<details>
  <summary>minimal XML</summary>

```XML
import argparse

import mujoco
import mujoco_viewer
def main(path: str = 'humanoid.xml'):
    model = mujoco.MjModel.from_xml_path(path)
    data = mujoco.MjData(model)
#physics = mujoco.Physics.from_xml_path(path)
    

    # create the viewer object
    viewer = mujoco_viewer.MujocoViewer(model, data)
    

    # simulate and render
    for _ in range(10000):
        if viewer.is_alive:
            mujoco.mj_step(model, data)
            #mujoco.mj_step(physics)
            viewer.render()
            print(data.geom_xpos)
            #print(physics.named.data.xpos)
        else:import argparse

    # close
    viewer.close()

if __name__ == "__main__":
    parser = argparse.ArgumentParser(description='Simulate')
    parser.add_argument('--path', type=str, help='Relative path to an XML file')
    args = parser.parse_args()

    if args.path is None:
        main()
    else:
        main(args.path)
```

_Originally posted by @makitayugo in https://github.com/deepmind/mujoco/issues/718#issuecomment-1425734847_

kevinzakka, 2023-02-10:
Hi @makitayugo, please don't duplicate issues for comments. Either continue the thread in the other issue, or make a discussion post.

closed 2023-02-10 by kevinzakka

yuvaltassa, 2023-02-10:
@makitayugo did you look at our [tutorial](https://github.com/deepmind/mujoco#getting-started)?
This is explained there very clearly. You want something like
```py
for i in range(model.ngeom):
  print(i, model.geom(i).name, data.geom(i).xpos)
```