google-deepmind/mujoco #732: Question regarding solref="−stiffness −damping"

emailweixu, 2023-02-13:
Hi,

I'm trying to understand the effects of solref. According to https://mujoco.readthedocs.io/en/latest/modeling.html#csolver,
when solref is in this format, $b=dampling/d_{max}$, $k=stiffness/d_{max}^2$. According to the equation of dynamics $a_1+d \cdot (bv+kr)=(1-d) \cdot a_0$, when the system reaches equilibrium (i.e. $a_1=0$, $v=0$), the contact residue is $r=\frac{(1-d)a_0}{d\cdot k}$. Plugging the above $k$, we get $$r = \frac{d\cdot(1-d)a_0}{stiffness} \, (1)$$
 
However, when I actually measured $r$ in the simulation, I found it's value is vastly different from what the above formula, 
instead it is more like $r=\frac{(1-d)a_0}{stiffness}$.

Here is what I did to measure $r$. I set the gravity to "0 0 -1" and let a ball of radius 1 drop from 2 meters high, wait until it stops moving and find out its final z coordinate. According to my understanding,  $r$ is $1 - z$. However, I found the $r$ values for different stiffness cannot be predicted by 

https://colab.research.google.com/drive/1dXSEVlg7SING50oZqW4srnuPknMG9ERG?usp=sharing

yuvaltassa, 2023-02-16:
Interesting. I see what you're asking and your math checks out at first glance, as does your code.
I couldn't figure it out quickly, but I did verify that the [source code](https://github.com/deepmind/mujoco/blob/1552b988ff697b6a25b315a3879046805c8b246d/src/engine/engine_core_constraint.c#L1136) matches the documentation.

Let me get back to you on this, in the meantime, see if you can spot something in the source link above.

emailweixu, 2023-02-16:
Thanks for pointing out the code. From the relevant code about $k$ below, it does seem to suggest documentation error.

https://github.com/deepmind/mujoco/blob/1552b988ff697b6a25b315a3879046805c8b246d/src/engine/engine_core_constraint.c#L1166-L1173

In the above code, for the case of (timeconst, dampratio), $k=1/(d_{max}^2\cdot timeconst^2 \cdot dampratio^2)$.
For the case of (-stiffness -damping), $k=stiffness/d_{max}^2$. Notice that $k$ for (timeconst, dampratio) is different from the documentation, which is $k=d(r)/(d_{max}^2\cdot timeconst^2 \cdot dampratio^2)$. I suspect that it is multiplied with $d(r)$ somewhere else in the code. It is natural to infer that for the case of (-stiffness, -damping), $k$ is also similarly multiplied with $d(r)$ at the same place.

yuvaltassa, 2023-12-12:
Took us a while to get to this 🙂 

> I suspect that it is multiplied with $d(r)$ somewhere else in the code.

Yes it is. Look at [this line](https://github.com/google-deepmind/mujoco/blob/aed4254d65d54d61a1d357df62c917ee4ec390a1/src/engine/engine_core_constraint.c#L2153). Here `KBIP[4*i]` is the stiffness without the $d(r)$ term. The $d(r)$ term is `KBIP[4*i+2]`, which was set on [this line](https://github.com/google-deepmind/mujoco/blob/aed4254d65d54d61a1d357df62c917ee4ec390a1/src/engine/engine_core_constraint.c#L1436).

However when investigating your question we realised that there is a bug in the documentation, the $d(r)$ term was missing from the direct formula. This will be fixed shortly, thanks for bringing it to our attention!

commit 64c33cef (2023-12-12): Correct formula for `solref` direct stiffness. Fixes #732.

PiperOrigin-RevId: 590144795
Change-Id: I94def777c7456243509374caecea2a335859f1f2
closed 2023-12-12 by copybara-service[bot]