google-deepmind/mujoco #733: How to stop/update the mujoco.set_mjcb_control()function ?

cidxb, 2023-02-13:
Hello.
Because I want to see how my robot moves, but the default viewer.launch will block the script process, so I took a look at the https://github.com/deepmind/mujoco/issues/649 and learned how to use mjcb_control to build a callback function to see my reboot execute my order, move to the position I want (the controlled actuators are the mujoco.data.ctrl).
I can see the movement, but now I want it to process a list of my order, I tried some thoughts, but it seems hard to get new input for the callback.

What I want is, when the reboot arrives at the first point, then it starts to move next point.

Here is the pseudo-code of my callback function:

def load_callback(self,m=None,d=None)
      mj.set_mjcb_control(None) 
     m=self.model
     d= self.data
  if m is not None:
     mj.set_mjcb_control(lambda m,d :self:move_group_to_joint_target_pos(m=m,d,=d,position))

The self:move_group_to_joint_target_pos is a function to set the data.ctrl

I have tried to write an if success arrive the position make the m=None to make it stop(  if m is not None:), but it seems not working,  the terminal report back that the None object can not  iterative 

THX !

cidxb, 2023-02-23:
Ok. Or maybe can give me a simple solution?

What I want is, when my RL algorithm/me giving the commands, I am able to see the according movements.
So I can adjust the model or the setting accordingly...

kbayes, 2023-02-24:
Hi cidxb,

I don't think you provided enough information for us to help you, but I'll give you my ideas on what's going wrong:

1.  The position variable might be ill-defined. In your code sample, position is never defined (I assume it's defined somewhere else), but it's being passed in your callback.
2.  In your last statement you mention something about setting m=None.  I don't know what your move_group_to_joint_target_pos callback looks like but you might be setting the local m to be None.  Do you want self.model to be None?  I can't tell what you're trying to do here.

cidxb, 2023-03-21:
> Hi cidxb,
>
> I don't think you provided enough information for us to help you, but I'll give you my ideas on what's going wrong:
>
> 1. The position variable might be ill-defined. In your code sample, [...]

Hey， thank you for your reply.
The original desire of my doing so is to have a passive viewer for watching my RL agent doing stuff, so I can monitor it.
So I want to use this callback function `mujoco.set_mjcb_control()` to set the action while I am using mujoco.viewer (), because of the mujoco.viewer will block the script.

But then I found the callback is a loop, once I give an order/action, then it will loop, and can't insert the next order. 

Instead, now I just use a CV window rendering the current scene with a fixed cam in mujoco.
Much easier.

I do see a post https://github.com/deepmind/mujoco/issues/766 ,  I think it might suggest the possible of a passive viewer in the future.

closed 2023-03-21 by cidxb

saran-t, 2023-03-21:
@cidxb The new `viewer.launch_passive` feature that we released yesterday (see also https://github.com/deepmind/mujoco/discussions/780) is exactly the non-blocking viewer that you were talking about in the top post.

cidxb, 2023-03-22:
> @cidxb The new `viewer.launch_passive` feature that we released yesterday (see also #780) is exactly the non-blocking viewer that you were talking about in the top post.

YES! I saw it yesterday too, thank you for your info on it!