google-deepmind/mujoco #739: ValueError: mass and inertia of moving bodies must be larger than mjMINVAL

mir7b, 2023-02-16:
Hello! 

While testing a composite body, adding the freejoint tag inside of the body causes the error `ValueError: Error: mass and inertia of moving bodies must be larger than mjMINVAL` . API reference notes "`mjMINVAL : 1E-15`", but I have:

```
<body name="A" pos="0.7 0.0 1.0">
    <freejoint />
    <inertial mass="0.001"/>
    <composite prefix="A" type="grid" count="16 16 1" spacing="0.005" flatinertia="0.01">
        <joint kind="main" damping="0.01"/>
        <geom type="capsule" size="0.004" priority="1"/>
    </composite>
</body>
```

Removing the freejoint kinda works, but then using `data.body("A").xpos` doesn't give the correct info. It just gives the original pos. (even after using `mujoco.mj_step(model, data)` ) 

I'll appreciate any advice on either how to get the correct pos at each timestep for this body, or how to fix the structure to get the freejoint working. 

---

On a side note, I have also tried using `type="cloth"` but that causes a different error: `ValueError: Error: sizes must be positive in object 'AG0_0' ` 
(even after fixing the `ValueError: XML Error: AB must be the beginning of root body name`) 

Thanks in advance!

mir7b, 2023-02-16:
Just realized that there is the capsule shape with wrong size structure for the cloth. So trying it as a cloth with either sphere as geom type or correct sizes for capsule works; although the body does not behave well as a cloth type (goes through a rigid box geom as if it's a ghost).

yuvaltassa, 2023-02-16:
Make sure you visualise the colliding geoms to understand what's going on.

To be clear, MuJoCo's soft body modeling can be much improved. We are working in this direction, hopefully with new soft body framework coming this year, but until then, I can't say that I recommend the current composite element. One should think about `composite` as "simple examples of procedurally constructed MuJoCo models". If I really wanted to simulate something like cloth, with MuJoCo's current abilities, I would use [PyMJCF](https://github.com/deepmind/dm_control/blob/main/dm_control/mjcf/README.md) to make my own composites that suite my use case. `<composite>` is just an MJCF macro (you tried loading and saving the model, yes?). You can make your own thing that will behave much better.

closed 2023-02-16 by yuvaltassa