google-deepmind/mujoco #741: Can we move a model in the environment using the keyboard

rajuptvs, 2023-02-19:
Hello everybody, 

We are trying to create an environment in which a model can be moved using the keyboard inputs (W,A,S,D), I am completely new to the mujoco, looked at the documentation but couldn't find any relevant info. 

Would be really helpful and appreciate if someone who has already worked on this could give us pointers on which documentation provides the info for using keyboard to move the model. 

Thanking you in advance

Balint-H, 2023-02-21:
Pranav Bhounsule's tutorial series covers interactive control of mujoco scenes with both [python](https://pab47.github.io/mujocopy.html) and [c/c++](https://pab47.github.io/mujoco.html) well. I'm not sure what is the best way to implement handling glfw events with the newer mujoco.viewer method. (E.g. lecture 7 for the python course)

If your project relies extensively on user interfacing, then I can also recommend considering the Unity plugin for developing interactive MuJoCo scenes.

rajuptvs, 2023-04-10:
@Balint-H  Thanks a lot, this worked. ,

closed 2023-04-10 by rajuptvs