google-deepmind/mujoco #742: Rendering cameras outside of the main thread on MacOS

aftersomemath, 2023-02-20:
It is straightforward on Windows and Linux to create cameras that can be rendered outside of the main thread. [Here](https://github.com/aftersomemath/mujoco/blob/simulate-python-examples/model/fixation/fixation.py) is an example of rendering a camera in the `on_control` callback of the physics thread of the Python interactive viewer.

However, on MacOS this approach fails. It is not straightforward to setup an OpenGL rendering context outside of the main thread. In order to support collaborators who primarily use MacOS, I'd like to find a workaround that can hopefully be merged with  MuJoCo or made into a standard example.

I've considered the following approaches:
- Find a way to render an OpenGL context on MacOS outside of the main thread. This appears to be [supported by Apple](https://developer.apple.com/library/archive/documentation/GraphicsImaging/Conceptual/OpenGL-MacProgGuide/opengl_threading/opengl_threading.html).
  - This may be as simple as using the current MuJoCo rendering API more carefully, I haven't checked the details yet
- Render the "extra" cameras inside of the C++ simulate thread and somehow make these frames available on other Python threads. This solution would have numerous technical limitations and would be specifically for Simulate only.

Is this a feature MuJoCo would be interested in supporting? I would be happy to help make it happen.

yuvaltassa, 2023-02-20:
Hi @aftersomemath !

Regarding option 2, we agree that this would be extremely cumbersome and does not seem like a good solution.

Regarding option 1, this is news to us! Many of the annoyances in the current viewer design have to do with this well-known MacOS limitation. If you can figure out a way to somehow circumvent this limitation (where the workaround is not worse than the current situation, natch), that would be fantastic. Please let us know if you make any progress in this direction, even if it is not a full submittable PR.

aftersomemath, 2023-02-21:
Great, I'll take a look into option 1.

I think the limitation with MacOS is not due to rendering but the way UI input events are handled on Mac. There is some explanation on the [GLFW forums](https://discourse.glfw.org/t/multithreading-glfw/573/5). Based on that, the current interactive viewer design must stay as is, but there may be hope for rendering cameras off the main thread.

saran-t, 2023-02-21:
I'm vaguely aware that the thread limitation relates to Cocoa UI handling rather than OpenGL, but I've been trying to avoid having to implement CGL-specific logic just to support offscreen rendering for macOS.

If you're volunteering though, feel free to dig into this :)

aftersomemath, 2023-02-22:
I realized that even if MuJoCo were to support offscreen rendering outside of the main thread on MacOS, it would still not be possible to show those cameras to the user because the non-main threads can't render to a window! (e.g. cv2.imshow would still not work: https://github.com/opencv/opencv/issues/22602). Showing images (and intermediate images) is essential to the computer vision workflow. So I found another way that allows both rendering and visualizing.

[Here](https://github.com/aftersomemath/mujoco_mouse_robot/blob/71ff7eb8330fdc7845bb3404efcfc5d040cd321c/mouse.py#L10) is a code that spawns a separate process that receives render requests and returns images over two multiprocessing Queues. I tested MacOS and Linux. Performance is ok, not great. The process must be "spawned" and not "forked" to avoid OpenGL issues.

I haven't been able to think of another solution that can visualize the rendered images in a flexible way on MacOS. So I think this is the best option for now.

aftersomemath, 2023-03-18:
It looks like the MacOS passive viewer introduced by 230e2780de5e3a973b79cb07b545da0764daf854 could more gracefully support this use case because it allows the user defined process to be the main process and the viewer to be the sub-process (communication still occurs over a Queue).

Interactivity could still work if somehow the viewer sent its state back over a second queue and `_physics_loop` in viewer.py is exposed to the user (they could also copy `_physics_loop` in order to keep interactivity and full control of the physics loop).

saran-t, 2023-04-07:
In theory, https://github.com/saran-t/mujoco/commit/e570cb57e8530d1d38990af9d572141acae4b54e should allow offscreen rendering from non-main thread on macOS (without the need for `mjpython` or anything special).

I don't suppose you're in the position to be able to help me verify?

aftersomemath, 2023-04-10:
It is difficult for me to test this right now as my collaborator with a Mac is fairly busy. The main example that I would try is [here](https://github.com/aftersomemath/mujoco/tree/simulate-python-examples/model/fixation). If the `cv2.imshow` calls are commented out it should just work if the CGL context is working.

saran-t, 2023-04-10:
I'm running into the following, which at first glance seems unrelated.

```Traceback (most recent call last):
  File "/Users/stunya/git/mujoco/model/fixation/fixation.py", line 100, in <module>
    viewer.launch(loader=load_callback)
  File "/Users/stunya/cgl/lib/python3.11/site-packages/mujoco/viewer.py", line 290, in launch
    _launch_internal(
  File "/Users/stunya/cgl/lib/python3.11/site-packages/mujoco/viewer.py", line 252, in _launch_internal
    simulate = Simulate()
               ^^^^^^^^^^
RuntimeError: Caught an unknown exception!```

saran-t, 2023-04-10:
Nvm, it's because `viewer.launch` doesn't work under `mjpython`.

Yes, the CGL logic seems to be working.

commit 6d63e046 (2023-04-10): Add low-level CGL context management for macOS.

This allows rendering from non-main threads as the context is not tied to a Cocoa window.

Fixes #742
Fixes #798

PiperOrigin-RevId: 523144359
Change-Id: I98432debdd034814c313eda94e38c248c3c1f76c