google-deepmind/mujoco #743: The depth/ distance in Mujoco

cidxb, 2023-02-21:
**How to ask for help**

First, read our quick guide to
[asking good questions](https://github.com/deepmind/mujoco#asking-questions).
Below is a template for you to use:

Hi,

I'm a (student / professor / engineer) and I'm trying to use MuJoCo for _____.

I'm looking for some help with ____.

Here is a model which explains my question:

<details>
  <summary>minimal XML</summary>

```XML
<mujoco>
  <worldbody>
    <light pos=".4 -.4 .3" dir="-2 2 -1.5" diffuse=".6 .6 .6"/>
    <light pos="-.2 -.4 .3" dir="1 2 -1.5" diffuse=".6 .6 .6"/>
    <geom type="plane" size=".5 .5 .01"/>
    <body name="ball" pos="0 0 0.05">
      <freejoint/>
      <geom size=".03"/>
    </body>
    <body name="hammer" pos="-.05 0 0.18">
      <joint axis="0 1 0"/>
      <geom type="capsule" size=".01" fromto="0 0 0 -.15 0 0"/>
      <geom type="box" size=".02 .02 .03" pos="-.15 0 0"/>
    </body>
  </worldbody>
</mujoco>
```

</details>

Here is a screenshot / video, illustrating my question:

yuvaltassa, 2023-02-21:
Why aren't you using the depth image functionality already provided?
This is described in our [tutorial](https://github.com/deepmind/mujoco#getting-started).
Note, this does **not** computed the distance from the position of the camera, but rather the z-depth, i.e. the distance from the plane of the camera.

closed 2023-02-21 by yuvaltassa

cidxb, 2023-02-23:
> Why aren't you using the depth image functionality already provided? This is described in our [tutorial](https://github.com/deepmind/mujoco#getting-started). Note, this does **not** computed the [...]

THX. but I did used the rendering method in the tutorial.
The depth image value is not the actual distance value. I want the actual distance...

yuvaltassa, 2023-02-23:
[Here](https://github.com/deepmind/dm_control/blob/774f46182140106e22725914aad3c6299ed91edd/dm_control/mujoco/engine_test.py#L66) is a _passing test_ of the same code, showing that the distance is correct.

Perhaps this will help you debug?

cidxb, 2023-03-24:
> [Here](https://github.com/deepmind/dm_control/blob/774f46182140106e22725914aad3c6299ed91edd/dm_control/mujoco/engine_test.py#L66) is a _passing test_ of the same code, showing that the distance is correct.
> 
> Perhaps this will help you debug?

Thank you!  I forget to show my progress.
After reviewing the code here and the instruction in the updated tutorial, I solved my problem.

There are two mistakes I made :
1. The depth image value is natively the depth value, and doesn't need to be processed (with that function  i`mage = near / (1 - self._depth_buffer * (1 - near / far))`)
2. There is a resize value of depth image operation in the tutorial, I check the tutorial a while before, and at that time there isn't a notation, and I thought that resize operation is a standard operation to get real value. Turns out it is just an example for a customized process of the depth image.
THX again!