google-deepmind/mujoco #744: Multiple view of cameras in a render window simultaneously.

joonhyung-lee, 2023-02-21:
Hi,

I'm a student and I'm trying to use MuJoCo for RL based robot control tasks .

I'm looking for some help about plot camera views.

I want to add camera image on render window. So, I wish the main camera and sub-camera could be simultaneously plotted in the window.
(I can do some simple add geoms. But I can't find add image or view in rendering window.)

Like the picture below in the gazebo.
Here is a screenshots in Gazebo, illustrating my question:

I'd really appreciate your help.

kevinzakka, 2023-02-22:
Hi @joonhyung-lee, this is definitely possible.

See this [side PR](https://github.com/kevinzakka/mujoco/blob/depth-sim/simulate/simulate.cc#L459) I worked on a few months ago to add a depth camera overlay in the simulate executable.

[image]

kevinzakka, 2023-02-22:
Side note: if you are using the Python bindings, you can simply do 2 calls to the render function with different `camera_id` values and potentially height/width values and then do stacking/overlaying logic in python.

joonhyung-lee, 2023-02-22:
Thank you very much for your answer with very detailed information.

I checked all the contents of the links. 
However, I wonder how I can perform the **stacking/overlaying** part in the render window using the python code.

I can get the depth image, and I wonder how I can **overlay** this image on rendering window. Could you help me?
(I know how to simply overlay the text, but I can't apply it to the image.)

kevinzakka, 2023-02-22:
@joonhyung-lee What rendering window are you using?

joonhyung-lee, 2023-02-22:
@kevinzakka I meant the render window that is output when using the mujoco library.

```
    every_second = 1 # (second)
    if (tick % int(every_second/dt/nstep)) == 0:
        img = np.zeros((viewer.viewport.height,viewer.viewport.width,3),dtype=np.uint8)
        depth_img = np.zeros((viewer.viewport.height,viewer.viewport.width, 1), dtype=np.float32)
        

        mujoco.mjr_render(viewer.viewport, viewer.scn, viewer.ctx)
        mujoco.mjr_readPixels(img, depth_img, viewer.viewport, viewer.ctx) # this makes a little delay
        
        img = np.flipud(img) # flip image
        depth_img = np.flipud(depth_img) # flip image

        imgs.append(img)
        depth_imgs.append(depth_img)
        img_ticks.append(tick)
```
[image]

closed 2023-02-23 by kevinzakka

dtorre38, 2024-02-15:
@jwcho0609 Thanks to kevinzakka's instructions I was able to get it working. Check out my simple implementation: https://github.com/dtorre38/mujoco_opencv.git

Muhammad540, 2024-10-21:
> I don't recognize that window, is that a third party viewer library? In any case, the code link I sent you has all the ingredients necessary to overlay a second camera view in this window.
> 
> 1. Create a rectangular viewport in the upper right corner for example.
> 2. Specify a different camera view by updating the scene with `mjv_updateScene`.
> 3. Render the scene in the offscreen buffer with `mjr_render`.
> 4. Read the pixels with `mjr_readPixels`.
> 5. Call `mjr_drawPixels` using the rectangular viewport you created in step 1.

Here i am sharing part of the code, that i wrote following the instructions to add an overlay cam by @kevinzakka. 
```python
# Create a viewport in the upper right corner
self.viewport_width, self.viewport_height = self.glfw.get_framebuffer_size(self.window)
camera_width, camera_height = 320, 240 
self.camera_buffer = mj.MjrRect(self.viewport_width - camera_width, self.viewport_height - camera_height, camera_width, camera_height)
# Create a separate camera object for rendering the sensor feed
self.cam_sensor_id = self.data.cam("add-your-cam-name").id
self.camera_sensor_view = mj.MjvCamera()
self.camera_sensor_view.fixedcamid = self.cam_sensor_id
self.camera_sensor_view.type = mj.mjtCamera.mjCAMERA_FIXED
# ...
# in the sim loop add this 
while not self.glfw.window_should_close(self.window):
            # ...
            # sim logic
            # ...
            # Render the camera view to the offscreen buffer (NOT the main simulation window)
            mj.mjv_updateScene(self.model, self.data, self.opt, None, self.camera_sensor_view, mj.mjtCatBit.mjCAT_ALL.value, self.scene)
            mj.mjr_render(self.camera_buffer, self.scene, self.context)

Muhammad540, 2024-10-21:
> I don't recognize that window, is that a third party viewer library? In any case, the code link I sent you has all the ingredients necessary to overlay a second camera view in this window.
>
> 1. [...]

Here i am sharing the code i wrote following @kevinzakka instructions to add a cam overlay.

# Create a viewport in the upper right corner
self.viewport_width, self.viewport_height = self.glfw.get_framebuffer_size(self.window)
camera_width, camera_height = 320, 240 
self.camera_buffer = mj.MjrRect(self.viewport_width - camera_width, self.viewport_height - camera_height, camera_width, camera_height)
# Create a separate camera object for rendering the sensor feed
self.cam_sensor_id = self.data.cam("add-your-cam-name").id
self.camera_sensor_view = mj.MjvCamera()
self.camera_sensor_view.fixedcamid = self.cam_sensor_id
self.camera_sensor_view.type = mj.mjtCamera.mjCAMERA_FIXED
# ...
# in the sim loop add this 
while not self.glfw.window_should_close(self.window):
            # ...
            # sim logic
            # ...
            # Render the camera view to the offscreen buffer (NOT the main simulation window)
            mj.mjv_updateScene(self.model, self.data, self.opt, None, self.camera_sensor_view, mj.mjtCatBit.mjCAT_ALL.value, self.scene)
            mj.mjr_render(self.camera_buffer, self.scene, self.context)

tenfoldpaper, 2024-11-07:
@dtorre38 Have you been able to fix the depth overlay issue? It seems that the current version of Mujoco does not support overwriting the rendered window's pixel values via `mjr_drawPixels`.

I copy-pasted @jwcho0609 's code into 3.2.5 simulate.cc, with the same placement and all, and then I try to set the value of the final `depth8` array all to 0, since that should render the small window to all black. But it seems that the window ignores everything after `mjr_render` is called.

I've made sure that there is no additional `mjr_render` calls that refers to this overlay viewport either.

```c++
// show depth overlay
void ShowDepth(mj::Simulate* sim, mjrRect rect) {
  // constant width with and without profiler
  int width = sim->profiler ? rect.width/3 : rect.width/4;

  // render figure in the top right
  mjrRect viewport = {
    rect.left + rect.width - width,
    rect.bottom + rect.height - rect.height/3,
    width,
    rect.height/2
  };

  /*
  // render scene in offscreen buffer
  */
  mjr_render(viewport, &sim->scn, &sim->platform_ui->mjr_context());
  
  int total = viewport.width*viewport.height;

  // read depth buffer, values are in range [0, 1]
  std::unique_ptr<float []> depth(new float[total]);
  mjr_readPixels(nullptr, depth.get(), viewport, &sim->platform_ui->mjr_context());

  // convert to meters
  float extent = sim->m_->stat.extent;
  float near = sim->m_->vis.map.znear * extent;
  float far = sim->m_->vis.map.zfar * extent;
  for (int i=0; i<total; ++i) {
    depth[i] = near / (1.0f - depth[i] * (1.0f - near/far));
  }

  // convert to a 3-channel 8-bit image
  std::unique_ptr<unsigned char[]> depth8(new unsigned char[3*viewport.width*viewport.height]);
  for (int i=0; i<viewport.width*viewport.height; i++) {
    unsigned char depth_value = (unsigned char)((1.0f - depth[i]) * 255.0f);
    depth8[3*i]     = depth_value;
    depth8[3*i + 1] = depth_value;
    depth8[3*i + 2] = depth_value;
    
  }
 // Simple test pattern (black image) for debugging
  // std::unique_ptr<unsigned char[]> depth8(new unsigned char[3 * viewport.width * viewport.height]);
  memset(depth8.get(), 0, 3 * viewport.width * viewport.height);  // Set all pixels to black

  // This does not do anything; original color image is rendered no matter what
  mjr_drawPixels(depth8.get(), nullptr, viewport, &sim->platform_ui->mjr_context());
}
```

dtorre38, 2024-11-07:
@tenfoldpaper My issue was solved via https://github.com/dtorre38/mujoco_opencv/issues/2#issuecomment-2239438569. It turns out that glClear(GL_DEPTH_BUFFER_BIT) must be called before mjr_drawPixels().

tenfoldpaper, 2024-11-08:
> @tenfoldpaper My issue was solved via [dtorre38/mujoco_opencv#2 (comment)](https://github.com/dtorre38/mujoco_opencv/issues/2#issuecomment-2239438569). It turns out that glClear(GL_DEPTH_BUFFER_BIT) must be called before mjr_drawPixels().

That's amazing, it worked without the hack!

For reference, in C++, this requires the following:

```cmake
# In the CMakeLists.txt for simulate
...
find_package(OpenGL REQUIRED)
...
# in the executable 
target_link_libraries(
    simulate
    libsimulate
    mujoco::mujoco
    glfw
    OpenGL::GL # <-- add this
    Threads::Threads
    lodepng
  )
...
```

```c++
// in the simulate.h header 
#include <GL/gl.h>
...
```

```c++
// in the simulate.cc, depthshow function:
  ...
  // render colorized depth
  glClear(GL_DEPTH_BUFFER_BIT);
  mjr_drawPixels(depth8.get(), nullptr, viewport, &sim->platform_ui->mjr_context());
 
```