google-deepmind/mujoco #745: On MacOS `mujoco.MjModel.from_xml_path` does not work with relative paths when used in the viewer loader function

aftersomemath, 2023-02-22:
On MacOS, a user defined "loader" function for the interactive viewer cannot use relative paths with `mujoco.MjModel.from_xml_path`. An "ERROR_FILE_NOT_FOUND" error occurs. Absolute paths work as expected.

Relative paths work with `mujoco.MjModel.from_xml_path` as long as it is called outside of the viewers loader callback.

A reproducing example is [here](https://github.com/aftersomemath/mujoco/blob/e1ba2e6e2220758539f034b5a5ddbcdfdb4c6e34/model/double_pendulum/double_pendulum.py#L58). Changing the path to an absolute path fixes the error.

nimrod-gileadi, 2023-02-28:
I just tried this on Linux, and it worked as expected, so this might be Mac specific.

kevinzakka, 2024-01-03:
Hi @aftersomemath, just tried this on my M1 and it works fine. Can you check again to see if it's been resolved in newer versions of MuJoCo?

aftersomemath, 2024-01-03:
It seems to be working now. I don't have access to a Mac personally, but @VAK1 ran it too and says its working as expected.

closed 2024-01-03 by aftersomemath