google-deepmind/mujoco #748: Geoms collision sliding and penetration

BaiYunpeng1949, 2023-02-23:
Hi,

I'm a student and I'm trying to use MuJoCo for building environments in reinforcement learning.

I'm looking for some help with how to disable the geom collision sliding and allow the geoms penetration without setting contact excludes.

Here is a model which explains my question:

<details>
  <summary>minimal XML</summary>

```XML
<?xml version="1.0"?>
<mujoco>
    <compiler angle="radian"/>
    <option collision="all" timestep="0.002"/>
    <size njmax="1000" nconmax="400"/>
    <default class="main">
        <default class="eye">
            <position ctrllimited="true" kp="5"/>
            <joint type="hinge" stiffness="0.0" damping="0.5" armature="0.0" limited="true"/>
        </default>
        <default class="grid">
            <geom type="box" size="0.0025 0.0001 0.0025" rgba="0.5 0.5 0.5 0.85" friction="1.0"/>
        </default>
    </default>
    <asset>
        <texture type="skybox" builtin="gradient" rgb1="1 1 1" rgb2="0.6 0.8 1" width="256" height="256"/>
        <texture type="cube" name="blue-eye" file="assets/blue_eye_texture_circle.png" gridsize="3 4" gridlayout=".U..LFRB.D.."/>
        <material name="blue-eye" texture="blue-eye" texuniform="true"/>
        <mesh name="eye" file="assets/basic_eye_2.stl" scale="0.05 0.05 0.05"/>
    </asset>
    <worldbody>
        <light name="scene_directional_light" directional="true" castshadow="false" cutoff="4" exponent="20"
               diffuse="1 1 1" specular="0 0 0" pos=".9 .3 2.5"/>

        <camera name="env" pos="-0.1 -0.05 0.05" euler="1.0 -0.85 -0.37"/>

        <body name="eye" pos="0 0 0" quat="0.7071068 0.7071068 0 0">
            <geom name="eye" type="sphere" size="0.025" material="blue-eye" rgba="1 1 1 1" quat="0.6625191 0.0130711 0.7488417 0.0115643"/>
            <camera name="eye" fovy="90"/>
            <joint name="eye-joint-x" class="eye" axis="1 0 0" range="-1.0 1.0"/>
            <joint name="eye-joint-y" class="eye" axis="0 1 0" range="-1.0 1.0"/>
            <site name="rangefinder-site" quat="0 0 1 0"/>

            <geom name="fixate-point" pos="0 0 -0.05" type="sphere" size="0.001" rgba="0.8 0.8 0 0.5" friction="1.0"/>
        </body>

        <body name="smart-glass-pane" pos="0 0.05 0">
            <geom name="grid-0" pos="-0.015 0 0.015" class="grid"/>
            <geom name="grid-1" pos="-0.005 0 0.015" class="grid"/>
            <geom name="grid-2" pos="0.005 0 0.015" class="grid"/>
            <geom name="grid-3" pos="0.015 0 0.015" class="grid"/>

            <geom name="grid-4" pos="-0.015 0 0.005" class="grid"/>
            <geom name="grid-5" pos="-0.005 0 0.005" class="grid"/>
            <geom name="grid-6" pos="0.005 0 0.005" class="grid"/>
            <geom name="grid-7" pos="0.015 0 0.005" class="grid"/>

            <geom name="grid-8" pos="-0.015 0 -0.005" class="grid"/>
            <geom name="grid-9" pos="-0.005 0 -0.005" class="grid"/>
            <geom name="grid-10" pos="0.005 0 -0.005" class="grid"/>
            <geom name="grid-11" pos="0.015 0 -0.005" class="grid"/>

            <geom name="grid-12" pos="-0.015 0 -0.015" class="grid"/>
            <geom name="grid-13" pos="-0.005 0 -0.015" class="grid"/>
            <geom name="grid-14" pos="0.005 0 -0.015" class="grid"/>
            <geom name="grid-15" pos="0.015 0 -0.015" class="grid"/>
        </body>

    </worldbody>
    <actuator>
        <position name="eye-x-motor" joint="eye-joint-x" class="eye" ctrlrange="-1.0 1.0"/>
        <position name="eye-y-motor" joint="eye-joint-y" class="eye" ctrlrange="-1.0 1.0"/>
    </actuator>
    <sensor>
        <rangefinder name="rangefinder" site="rangefinder-site"/>
    </sensor>
    <contact>
        <!--        <exclude body1="eye" body2="smart-glass-pane"/>-->
    </contact>
</mujoco>
```

</details>

Here is a screenshot / video, illustrating my question:

https://user-images.githubusercontent.com/54785878/220900453-958cfef1-7fd7-4f3e-9bf0-63b632dec76e.mp4

<br />The **objective** of the task is using a rangefinder sensor to detect which cell is being focused by the eye, i.e., which is the closest cell being touched by the ray shooting out of the eye. We used a tricky method: we place the "fixate-point" sphere on the focused cell by changing its z-axis position. Then we use a collision/contact detection in mujoco.data to find the cell id by geom2:
`
#### Update fixate point based on rangefinder
        x = self._data.sensor("rangefinder").data
        x = x if x >= 0 else self._rangefinder_cutoff
        self._model.geom("fixate-point").pos[2] = -x
        if x != self._rangefinder_cutoff and len(self._data.contact.geom2) > 0:
           geom2 = self._data.contact.geom2[0]

However, as shown in the video, the placed fixate-point geom will collide and slide on the cell to avoid penetration, without specifying the contact excluding, no penetration is allowed. 

My question is: is there any other methods except for excluding contacts that allows no-sliding collisions / penetration? Since I still want to use the contact.geom2 to detect the focus target. And is [mj_ray](https://mujoco.readthedocs.io/en/stable/APIreference.html#mj-ray) in [Ray Collisions](https://mujoco.readthedocs.io/en/stable/APIreference.html#ray-collisions) a solution? Can you provide me some examples of using that if possible? And is there any other better ways to detect the eye's focus point?

Thank you for your time and support!

Best regards
Bai

nimrod-gileadi, 2023-02-27:
Hi Bai,
That's a nice creative solution, but I think that for this use case, you're better off using `mj_ray` directly, rather than using a rangefinder sensor.

You can look at the implementation of the rangefinder sensor [here](https://github.com/deepmind/mujoco/blob/5e989d18c349e914fb82ae937006ff4832b011ce/src/engine/engine_sensor.c#L224) to get the details.

I would be something along the lines of:
```
def ray_from_site(model, data, site_name="rangefinder-site"):
  site = data.site(site_name)
  pnt = site.xpos
  vec = site.xmat.reshape((3,3))[:, 2]
  # exclude the body that contains the site
  bodyexclude = model.site_bodyid[site.id]
  geomid_out = np.array([-1], np.int32)
  distance = mujoco.mj_ray(
      model, data, pnt, vec, geomgroup=None, flg_static=1,
      bodyexclude=bodyexclude, geomid=geomid_out)
  return distance, geomid_out[0]
```

BaiYunpeng1949, 2023-03-01:
Thank you for the help! 

Update:
The uncontrolled jitters are basically removed when I directly use mj_ray to detect the focus point.