google-deepmind/mujoco #750: Penetration occurs when two "box" type geoms contact

Rancho-zhao, 2023-02-26:
Hi,I'm a student and I'm trying to use MuJoCo for  peg-in-hole task.

There are two objects in my scene, including a cube and a box with a hole, which are both constructed using the "box" type geoms.

However, when the cube contacts the hole box, I find that a penetration issue occurs just like the following figure:

[image]

Additionally, I use mocap to control the motion of robot arm.

I have tried to change the value of "condim" and enable "multiccd", but it doesn't seem to work.

Here is a model which explains my question:

<details>
  <summary>minimal XML</summary>

```XML
<?xml version="1.0" encoding="utf-8"?>
<mujoco>
	<compiler angle="radian"  coordinate="local" meshdir="../stls/robot_tactile" ></compiler>
	<option timestep="0.002" cone="elliptic">
		<flag warmstart="enable" multiccd="enable"></flag>
	</option>
    <!-- <option timestep="2e-3"
            iterations="100"
            tolerance="1e-10"
            impratio="10"/> -->
    <default>
        <light castshadow="false" diffuse="1 1 1" />
        <camera fovy="60" />
        <motor ctrllimited="true" ctrlrange="-0.04 0.04"/>
        <!-- <joint range="-1.5708 1.5708"/> -->
    </default>

    <visual>
        <rgba connect="1 0.2 1 1" joint="0.2 0.6 0.8 1"/>
        <map fogstart="3" fogend="5"/>
        <quality shadowsize="2048"/>
    </visual>
    <option gravity="0 0 -9.81" iterations="100"/>

    <custom>
        <numeric name="N_GRIPPER_JOINTS" data="0" />
    </custom>

    <size nconmax="2000" njmax="10000" nstack="500000"/>

    <contact>
        <pair geom1="object0" geom2="left"/>
        <pair geom1="object0" geom2="right"/>
        <pair geom1="object0" geom2="front"/>
        <pair geom1="object0" geom2="back"/>
    </contact>
	
	<worldbody>

		<include file="robot.xml"></include>
            <body name="object0" pos="0 0.177 0.093" euler="0 1.5708 -1.5708">
                <site name="object0:site" pos="0 0 0" rgba="0 1 0 .1" size="0.02 0.02 0.02"/>
                <geom name="object0" type="box" size=".02 .02 .02" mass="1" contype="1" conaffinity="32" condim="3" rgba="1 1 0 1" />
            </body>	 
        <body name="object1" pos="-0.35 0.1 .62">
            <site name="object1:site" pos="0 0 0" rgba="0 0 0 0" size=".02 .02 .02"/>
            <geom name="left" type="box" pos="-.041 0 .02" size=".02 .06 .02" mass="1" contype="32" conaffinity="32" condim="3" rgba=".5 .5 .5 1"/>
            <geom name="right" type="box" pos=".041 0 .02" size=".02 .06 .02" mass="1" contype="32" conaffinity="32" condim="3" rgba=".5 .5 .5 1"/>
            <geom name="front" type="box" pos="0 .041 .02" size=".06 .02 .02" mass="1" contype="32" conaffinity="32" condim="3" rgba=".5 .5 .5 1"/>            
            <geom name="back" type="box" pos="0 -.041 .02" size=".06 .02 .02" mass="1" contype="32" conaffinity="32" condim="3" rgba=".5 .5 .5 1"/>
        </body>

        <light directional="true" ambient="0.2 0.2 0.2" diffuse="0.8 0.8 0.8" specular="0.3 0.3 0.3" castshadow="false" pos="0 0 4" dir="0 0 -1" name="light0"></light>
	</worldbody>
    <!-- Camera settings -->
    <visual>
        <!-- allows the camera to fit into smaller spaces without clipping through -->
        <map znear="0.001" zfar="1000"/>
    </visual>
    <statistic extent="1"/>

</mujoco>

```

</details>

yuvaltassa, 2023-02-28:
If you visualise contacts and/or contact forces do you see them?
i.e. are contacts happening and having no effect or are there no contacts?
@nimrod-gileadi can contype/affinity even be set to 32?

Rancho-zhao, 2023-02-28:
The contact forces are like these:
 
[image]
[image]

yuvaltassa, 2023-02-28:
Looking at your XML, your body appears to have no DoFs (joints), so it cannot move.

closed 2023-02-28 by yuvaltassa

Rancho-zhao, 2023-02-28:
yes, I need object1 stays fixed in my scene, what you said above means if I don't add any joint in body, penetration is certain to occur when it's too much contact?

yuvaltassa, 2023-02-28:
More like, if there's penetration, there's no way for it to be resolved. The forces are pushing on something that cannot move.