google-deepmind/mujoco #751: free joint but use euler angle for body orientation

rongakowang, 2023-03-01:
The current setting for the <freejoint /> uses quaternion for orientation, such that model.nq != model.nv which causes a lot of inconveniences, e.g. the gradient with respect to the rotation does not match the rotation shape so that we can't do a gradient descent optimization using the calculated gradient. I wonder if there is a compiler arg to allow free joints to use 3 dof rotation (e.g. euler angle) so that I don't need to create dummy bodies that possess some mass or inertia?

yuvaltassa, 2023-03-01:
No, there is no such setting. Using Euler angles for free joints is strongly discouraged as gimbal lock will create artifacts. Unless your rotations are bounded, don't do this.

The gradients do match, just not in dimension. Do you need help understanding how to use quaternion derivatives? If you explain your use case better, I can try to help.

closed 2023-03-01 by yuvaltassa

rongakowang, 2023-03-01:
@yuvaltassa Thank you for your explanation, I am doing normal gradient descent optimization for the quaternion `q`, for the derivative of `L` relative to `q`,  then this derivative will be of shape nv not nq, how should I perform q - lambda * dL/dq for some learning rate  lambda?

yuvaltassa, 2023-03-02:
You should use `mj_integratePos` and `mj_differentiatePos`, or if working just with the quaternion, `mj_quatIntegrate` and `mj_subQuat`/`mj_quat2Vel`. Take a look at the code implementations of these functions and/or look at our [LQR tutorial](https://github.com/deepmind/mujoco#getting-started), which uses these.