google-deepmind/mujoco #753: Creating / Destroying geoms during simulation

SimiPixel, 2023-03-01:
Hi,

I'm a PhD student and I'm trying to use MuJoCo for Machine Learning based Controller Design.

Is is possible to create geoms that (dis-) appear at some timepoint? (or via a programmatic interface via python *during* a mujoco simulation).

It would be very helpful to include geoms that highlight reference positions, but those reference positions can change over time.

Thanks for your time! Greetings
Simon

yuvaltassa, 2023-03-01:
Do you mean that you want to draw 3D objects in the visual scene? This is easy, there is an example in the tutorial colab.

Or do you mean that you want to change the physical model at runtime? This is not supported, but there are workarounds.

SimiPixel, 2023-03-01:
I meant that the `reference` geom's e.g. x position changes as function of time. So, here it is at `pos_x=0.5` initially, but after say five seconds the reference position has changed to `pos_x=1.5`

```xml
<mujoco>
  <worldbody>
  <geom name="reference" type="box" size=".2 .2 .2" rgba="1 0 0 1 pos=".5 0 0"/>
  </worldbody>
</mujoco>
```

yuvaltassa, 2023-03-01:
That doesn't answer my question.

Do you want a purely visual geom or do you need it to collide or otherwise participate in the physics?

SimiPixel, 2023-03-02:
Ah i see, i didn't realize what you meant by "physical model". 
No, the geom is purely illustrative / visual, and i can guarantee that nothing physical will interact (collide) with it.

yuvaltassa, 2023-03-02:
So add geoms to the scene, like in the tutorial.

closed 2023-03-02 by yuvaltassa

yuvaltassa, 2023-03-09:
If you want change the model between episodes, [PyMJCF](https://github.com/deepmind/dm_control/blob/main/dm_control/mjcf/README.md) offers a good solution. There are many examples for this in the dm_control repo.

SimiPixel, 2023-03-10:
To give others a starting point, i achieved my goal with the following snippet

```python3

from dm_control import mujoco
physics = mujoco.Physics.from_xml("...")

# this function is from the tutorial notebook
def add_visual_capsule(scene, point1, point2, radius, rgba):
    """Adds one capsule to an mjvScene."""
    if scene.ngeom >= scene.maxgeom:
        return
    scene.ngeom += 1  # increment ngeom
    # initialise a new capsule, add it to the scene using mjv_makeConnector
    mujoco.mjv_initGeom(
        scene.geoms[scene.ngeom - 1],
        mujoco.mjtGeom.mjGEOM_CAPSULE,
        np.zeros(3),
        np.zeros(3),
        np.zeros(9),
        rgba.astype(np.float32),
    )
    mujoco.mjv_makeConnector(
        scene.geoms[scene.ngeom - 1],
        mujoco.mjtGeom.mjGEOM_CAPSULE,
        radius,
        point1[0],
        point1[1],
        point1[2],
        point2[0],
        point2[1],
        point2[2],
    )

def make_scene_callback(pos_fn, rgba):
    def scene_callback(physics, scene) -> None:
        point1 = pos_fn(physics.time())
        point2 = point1 + np.array([0.0, 0, 0.01])
        add_visual_capsule(scene, point1, point2, 0.15, rgba)
    return scene_callback

here = np.array([0.0, 0.0, 0.0])
there = np.array([1.0, 0.0, 0.0])
rgba_blue = np.array([0.012, 0.263, 0.8745, 1.0])
scene_callback = make_scene_callback(lambda t: here if t<5 else there, rgba_blue)

T = 10.0
frames = []
while physics.time() < T:
    physics.step()
    frames.append(physics.render(scene_callback=scene_callback)
```