google-deepmind/mujoco #755: Object File Assumed

Eetana, 2023-03-03:
Hi,

I'm a student and I'm trying to use MuJoCo for a project.

I'm looking for some help with an error I’m encountering. The error occurs when I try using the “make” line command, and I get the following 
“mujoco-2.3.2-windows-x86_64\sample>make
cl /O2 /MT /EHsc /arch:AVX /I../include /Fe../bin/ testxml.cc                                ../lib/mujoco.lib
Microsoft (R) C/C++ Optimizing Compiler Version 19.35.32215 for x64
Copyright (C) Microsoft Corporation.  All rights reserved.

cl : Command line warning D9024 : unrecognized source file type 'C:/MinGW/msys/1.0/O2', object file assumed
cl : Command line warning D9024 : unrecognized source file type 'C:/MinGW/msys/1.0/MT', object file assumed
cl : Command line warning D9024 : unrecognized source file type 'C:/MinGW/msys/1.0/EHsc', object file assumed
cl : Command line warning D9024 : unrecognized source file type 'C:/MinGW/msys/1.0/I../include', object file assumed
cl : Command line warning D9024 : unrecognized source file type 'C:/MinGW/msys/1.0/Fe../bin/', object file assumed
testxml.cc
testxml.cc(20): fatal error C1083: Cannot open include file: 'mujoco/mjxmacro.h': No such file or directory
make: *** [all] Error 2

I get the same result using only x64 and x86-64 toolkits.
All help will be appreciated!

saran-t, 2023-03-08:
For some reason the compiler is treating those flags as file paths. I've never seen this before but hopefully that helps you troubleshoot the issue?

Also, our primary supported build tooling is CMake. The Makefiles are unmaintained and only provided as an example (which will probably be removed at the next release).

Please consult our documentation for build instructions.

saran-t, 2023-03-16:
I'm going to close this issue, but if you have any further info/question please feel free to reopen.

closed 2023-03-16 by saran-t