google-deepmind/mujoco #756: MuJoCo Mass & Collisions Problem

jmeredith20, 2023-03-04:
Hi all, I am trying to create a soft surface in MuJoCo such that the larger the mass of an object is, the more it should sink into the floor (like how if you put a feather on a trampoline it won't really sink into it at all, but if you put a weight on a trampoline then the weight would sink a bit into the trampoline and stay sunken down). I am using MuJoCo's solref parameters to attempt to do this. I tried applying solref to the ground itself and dropping a rigid block onto that ground, and then separately tried pairing the ground to a block I drop onto the ground with solref. With both approaches, increasing acceleration due to gravity affects how much the block will sink into the surface and changes its final position accordingly. However, no matter how much I increase or decrease the mass of the block being dropped onto the surface (while keeping gravitational acceleration constant), there is no difference in the final displacement value of the block into the surface. Does anyone have any idea how I can fix this and make it so the system recognizes that a higher mass means that the block should be sinking deeper? (This project is being done using Python and .xml files with MuJoCo). Thus far I have tested defining the mass of the dropped block in the geom parameter and the inertia parameters, both of which have not worked. Thank you in advance!!

Balint-H, 2023-03-07:
Depending on your application (namely if there's just a single object interacting with the ground) you could get away by adding a vertical sliding DoF to the ground with a high stiffness and higher damping. To render it as sinking into the ground, you can set the real ground transparent, and add a non-colliding reference geom as the ground visualization.

yuvaltassa, 2023-03-08:
If you want to affect the final penetration of contacts, you should be modifying `solimp`, not `solref`.

Play with this parmeter and let us know how it goes.

closed 2023-03-08 by yuvaltassa