google-deepmind/mujoco #757: How to wrap a rope around a pulley and make it turn in Mujoco?

TS35, 2023-03-05:
**How do I wrap a rope around a pulley and make it turn?**

Hi,

I wonder how to wrap a rope around a pulley (the second joint in the XML file below) and make the pulley (the second joint in the XML file below) turn in Mujoco?

Here is a model which explains my question:

<details>
  <summary>minimal XML</summary>

```XML
<mujoco model="test">
    <option gravity="0 0 -9.81" />
    <compiler coordinate="global"/>

    <default>
        <geom rgba=".9 .7 .1 1" size="0.01"/>
        <site type="sphere" rgba=".9 .9 .9 1" size="0.005"/>
        <joint type="hinge" axis="0 1 0" limited="true" range="0 60" solimplimit="0.95 0.95 0.1"/>
    </default>
    <asset>
        <material name="test_color" texture="checks1" texrepeat="2 2" texuniform="true"/>
        <texture name="checks1" builtin="checker" type='2d' width='256' height='256' rgb1="1 1 1" rgb2="0 0 0" />
    </asset>
    <visual>
        <headlight diffuse=".7 .7 .7"/>
    </visual>

    <worldbody>
        <body>
            <geom type="cylinder" fromto="-0.03 0 0.2  -0.03 0 0.15" size="0.03" rgba=".2 .2 .5 1" density="6000"/>
            <joint name="t1" type="slide" pos="-0.03 0 0.2" axis="0 0 1" limited="false"/>
            <site name="s1" pos="-0.03 0 0.2"/>
        </body>

        <site name="s2" pos="-0.03 0 0.32"/>

        <body>
            <geom name="g1" type="cylinder" fromto="0.0 0.015 0.3  0.0 -0.015 0.3" size="0.02" rgba=".3 .9 .3 .4" material="test_color"/>
            <joint name="Pulley" pos="0 0 0.3" frictionloss="0" damping="0"/>
            <site name="side1" pos="0 0 0.33"/>
        </body>
        <body name="tes" pos="0.05 0 0.32" >
            <geom type="sphere" size="0.03" rgba=".2 .2 .5 1" density="10000"/>
            <joint name="t3" type="slide" pos="0.05 0 0.3" axis="0 0 1" limited="false"/>
            <site name="s3" pos="0 0 0"/>
        </body>
    </worldbody>
    <tendon>
      <spatial width="0.002" rgba=".95 .3 .3 1" limited="true" range="0 5" frictionloss="100" damping="100">
            <site site="s1"/>
            <site site="s2"/>
            <geom geom="g1" sidesite="side1"/>
            <site site="s3"/>
      </spatial>
    </tendon>
</mujoco>
```

</details>

Here is a screenshot / video, illustrating my question:
[image]
[image]

https://user-images.githubusercontent.com/126997839/222938428-6d0e5501-d4a8-4e81-83e5-43ba06fd1f7f.mp4

As shown in the video, when the "s3" site falls with the rope, the black and green pulleys do not turn with it.
I thought that as the “s3” site falls with the rope, the pulley should rotate with the pull of the rope and friction, and I didn't know how to solve this problem.

yuvaltassa, 2023-03-05:
There is no friction force between tendons and wrapping geoms. They are idealised obstacles that only change the path of the tendon.

If you search the issues for winch you find one with an example of a physical rope wrapping around a winch, which does what you expect, but this is far more expensive and less stable.

closed 2023-03-05 by yuvaltassa

HILMR, 2023-04-08:
@yuvaltassa Sorry to bother you, I have reviewed your response to this issue, but I still have confusion. [The document](https://mujoco.readthedocs.io/en/latest/XMLreference.html#tendon) mentions that tendons can be used to impose dry friction forces, and [spatial tendons](https://mujoco.readthedocs.io/en/latest/XMLreference.html#tendon-spatial)  also provide the **frictionloss** parameter, what is the purpose of this parameter?

yuvaltassa, 2023-04-08:
This is friction **along the tendon**. Imagine two string wrapped together, such that that when a force greater than F is applied they start to slide apart.

HILMR, 2023-04-08:
> This is friction **along the tendon**. Imagine two string wrapped together, such that that when a force greater than F is applied they start to slide apart.

Thank you, I see, but still hope to achieve friction between the tendons and the geoms. The tendon part is the most attractive part of Mujoco to me!