google-deepmind/mujoco #759: Error: eigenvalue of mesh inertia too small

quangminh2702, 2023-03-07:
Hi,

I'm using MuJoCo for simulation of an assembly process. I decomposed an object into a union of few hundreds convex geoms using V-HACD and then using obj2mjcf to generate a XML file for that object.

When dropping that XML model into MJC 2.10, there exists an **_error: eigenvalue of mesh inertia too small_**

Is there any way to deal with this other than scaling up the mesh or reduce the number of decomposed geoms?

Thank you for your support!

yuvaltassa, 2023-03-07:
Do the compiler attributes boundmass and boundinertia not fix this?

That said, this implies a malformed mesh, possibly coplanar vertices...?

Try to examine the specific offending mesh. If it looks normal please attach it here, that could imply a bug.

quagla, 2023-03-09:
Also, does the error persist if you use `shellinertia="true"` in the compiler flags?

quagla, 2023-03-15:
@quangminh2702 Closing it on the assumption that the mesh was indeed malformed. Feel free to reopen if you think there is an issue.

quangminh2702, 2023-03-20:
Thank you for your replies.
I have tried using the compiler attributes _boundmass_ and _boundinertia_, but they does not fix the issue. When I tried `shellinertia="true"`, it showed `XML Error: Schema violation: unrecognized attribute: 'shellinertia'`.

The mesh file can be opened in Meshlab, and it works fine in Mujoco if I scaled it up to, for example, `scale="1000 1000 1000"`
I would like to attach the mesh file [here](https://drive.google.com/file/d/1qLcBpxa_dsPXK6IJuEVInJoS2Alz4T8s/view?usp=sharing) for your references. Thank you!

yuvaltassa, 2023-03-20:
@quagla made a misleading recommendation.

- shellinertia is [an attribute of geom](https://mujoco.readthedocs.io/en/latest/XMLreference.html#body-geom-shellinertia).
- exactmeshinertiais [an attribute of compiler](https://mujoco.readthedocs.io/en/latest/XMLreference.html#compiler-exactmeshinertia).

quangminh2702, 2023-03-21:
Thanks for your reply, @yuvaltassa .  Including `shellinertia` doesn't work either. Could you please try the [mesh file](https://drive.google.com/file/d/1qLcBpxa_dsPXK6IJuEVInJoS2Alz4T8s/view) and/or reopen the issue? Thanks!

quagla, 2023-03-21:
If you use both flags at the same time, then it works (apologies that I missed about mentioning both). However, the mesh is not rendered so my guess is that all faces have inverted (clockwise) orientation. Could you please double check that?

closed 2023-03-24 by quagla

quangminh2702, 2023-03-28:
Hi @quagla. As for the mesh file, I think there is no issue with it. If I scale the mesh file up using `scale="1000 1000 1000"`, then it becomes big enough to avoid the small eigenvalue issue of mesh inertia. It can be rendered and seen as in Mujoco as followings
[image]
(I also checked the mesh file using Blender to make sure that no face has inverted orientation)

However, we would like to work with the original size of the mesh file. I have tried to include both flags (`boundmass` and `boundinertia` in `compiler` and `shellinertia` in `geom`) at the same time, but still there is an error `XML Error: Schema violation: unrecognized attribute: 'shellinertia'`.  If I removed `shellinertia="true"`, then **error: eigenvalue of mesh inertia too small** persists.
Here are the [XML file](https://drive.google.com/file/d/1Vb123lKsztNhWLeIY70mKi7ER6IEDGUO/view?usp=sharing) and the [mesh file](https://drive.google.com/file/d/1qLcBpxa_dsPXK6IJuEVInJoS2Alz4T8s/view) that I work around. **Would it be anything that I made wrong there in the XML file?** If it worked for you last time, then I guess it was actually rendered, but it was too small and only difficult to find out.

quangminh2702, 2023-03-30:
Hi @quagla and @yuvaltassa. Could you please help reopen the issue so I hope that someone could be able to help?
Thank you so much.

commit c9437975 (2023-04-17): Make `shellinertia` ignore `exactmeshinertia`, which is only used for legacy volume computations. Fixes #759.

PiperOrigin-RevId: 524884880
Change-Id: Ia5b64ee33802866f792cdad4acfec6d9baf2a9e9