google-deepmind/mujoco #762: About the camera intrinsics in Mujoco

Lucky-hqj, 2023-03-08:
I want to set the camera intrinsics in Mujoco. How can I do that?
For example, these parameters like this:     intrinsics = (450., 0, 320., 0, 450., 240., 0, 0, 1)
Furthermore, How should I use the mjvGLCamera in python api ?

yuvaltassa, 2023-03-08:
Cameras are modifiable via [XML](https://mujoco.readthedocs.io/en/latest/XMLreference.html#body-j-camera). Does this give you enough control over intrinsics?

If not, you should be able to edit the glCamera directly by writing into the [frustum](https://mujoco.readthedocs.io/en/latest/APIreference/APIdatastructures.html#mjvglcamera). This will give you more control, but still not allow you to introduce skewed or trapezoidal images.

How much control do you need, and why?

Lucky-hqj, 2023-03-09:
I'm sorry, I had some confusion earlier. I have a new question now. When I use camera mode "targetbody", how can I directly
 get the rotation matrix of the camera from mujoco? I found that using "targetbody" method, model.cam_quat returns the default value, rather than the actual orientation.

DavidPL1, 2023-03-09:
> model.cam_quat returns the default value, rather than the actual orientation.

Information found in the model should be set only once: when a model is loaded. So this makes perfect sense. 

In targetbody mode the camera orientation does change depending on the current timestep.
You do have `data.cam_xmat` available to track this change, which should provide you the current orientation in cartesian space. 
I do think, however, that an updated relative orientation in form of a quaternion (e.g. `data.cam_quat`) would be nice to consistently keep track of (trees of) relative transformations.

Lucky-hqj, 2023-03-09:
I want to reconstruct a point cloud from RGB images and depth images. Can you give me some advice? I found the depth images in Mujoco seem unusual, how can I convert them to distances?

Lucky-hqj, 2023-03-09:
> I want to reconstruct a point cloud from RGB images and depth images. Can you give me some advice? I found the depth images in Mujoco seem unusual, how can I convert them to distances?

Sorry, the depth map is not the problem. But I am confused about the data.cam_xmat function, which seems to be inconsistent with the value I calculated. When using it to reconstruct the point cloud, I always fail to get the correct result.

Lucky-hqj, 2023-03-09:
> Information found in the model should be set only once: when a model is loaded. So this makes perfect sense.
>
> In targetbody mode the camera orientation does change depending on the current timestep. [...]

I am confused about the data.cam_xmat function, which seems to be inconsistent with the value I calculated. When using it to reconstruct the point cloud, I always fail to get the correct result.

DavidPL1, 2023-03-10:
I have currently no time to double check the values of `data.cam_xmat`.

Does your point cloud projection work with the initial state (without moving forward in time) when you use `model.cam_quat`? 
If not, keep in mind that the actual camera orientations are rotated relative to their body frame (quoting the [documentation](https://mujoco.readthedocs.io/en/stable/XMLreference.html#body-camera)):

> The camera is looking along the -Z axis of its frame. The +X axis points to the right, and the +Y axis points up.

yuvaltassa, 2023-03-12:
@DavidPL1's comments are all correct, thanks David!

I would also add that in MuJoCo, rotation matrices are such that the unit vectors lie on the **columns**. In other words, consider the x-y-z (red-green-blue) frame visualization cylinders that you can see in `simulate`, the vectors corresponding to these cylinders, in the world frame, would be `x  = {xmat[0], xmat[3], xmat[6]}` etc.

closed 2023-03-12 by yuvaltassa