google-deepmind/mujoco #764: Is it legit to update `body_mass` during simulation without resetting the whole simulation?

breakds, 2023-03-10:
Hi the great Mujoco community! I have 2 questions about updating `body_mass` of the `mjModel`.

1. If the `body_mass` of a specific body in a Mujoco model is updated in the middle of the simulation followed with `mj_forward` function (similar to what `dm_control` is doing), and then the simulation is continued using the `mj_step` function, will the updated mass value be taken into account in the subsequent simulation steps?

2. Does the `body_subtreemass` field in a Mujoco model participate in the simulation, and what happens if it is not updated when the `body_mass` of a body is updated?

Without digging into the implementation, my preliminary understanding is that if I update the `body_mass` of a specific body, I should also update the `body_subtreemass` of all its ancestors in the model tree to ensure that the mass distribution of the system remains consistent. If I do not update the `body_subtreemass `of the ancestors, it may cause inconsistencies in the computation of the dynamics quantities and lead to incorrect simulation results. 

Can someone with more experience help cofirm that?

Thanks!

yuvaltassa, 2023-03-10:
1. Yes. If you update `body_mass` that will be taken into account subsequently.
2. The most important additional value, which certainly participates in the simulation, is `body_inertia`. If you changed the mass from A to B, then you need to scale the inertia by B/A.
3. To be honest, I'm not sure about `body_subtreemass`, and perhap there are other fields that need to be changed. My instinct regarding body_subtreemass` is that it only participates in sensor calculations like `subtreecom` etc. The way to check this is to compare two models, one by changing the mass "properly" (via XML), and the other by changing the `mjModel`. If changing mass and inertia gives you the same results as changing the XML then you're done.

closed 2023-03-10 by yuvaltassa

breakds, 2023-03-10:
Thanks a lot for the prompt response, @yuvaltassa. I definitely missed `body_inertia`.

breakds, 2023-03-11:
I've compared the two approaches in No.3 as suggested by @yuvaltassa. In my case the following fields of `mjModel` are also different:

1. `actuator_acc0`
2. `dof_invweight0`
3. `dof_M0`
4. `body_invweight0`

Are they going to make a difference in the subsequent simulation?

Thanks

breakds, 2023-03-11:
Okay, I think I found it out. Calling `mj_setConst` does the trick. It will make the models identical in No.3 as suggested by @yuvaltassa