google-deepmind/mujoco #765: OpenGL_accelerate.errorchecker._ErrorChecker.glCheckError OpenGL.raw.EGL._errors.EGLError: <exception str() failed>

cidxb, 2023-03-13:
Hi, friends.

I have run into a problem with OpenGL.
So at first, I have the 

```
self._mjr_context = _render.MjrContext(
mujoco.FatalError: gladLoadGL error
```

So I have used the  command 
export MUJOCO_GL=egl

It solves the problem, and the script works.

But the error became 
```
exception ignored in: <function GLContext.__del__ at 0x7f46c576f250>
Traceback (most recent call last):
  File "/home/jeffery/anaconda3/envs/RL-server/lib/python3.10/site-packages/mujoco/egl/__init__.py", line 130, in __del__
  File "/home/jeffery/anaconda3/envs/RL-server/lib/python3.10/site-packages/mujoco/egl/__init__.py", line 123, in free
  File "/home/jeffery/anaconda3/envs/RL-server/lib/python3.10/site-packages/OpenGL/error.py", line 230, in glCheckError
OpenGL.raw.EGL._errors.EGLError: <exception str() failed>
```

and while I  install PyOpenGL-accelerate, it change to (guess is the same)
```
Exception ignored in: <function GLContext.__del__ at 0x7f1c14000dc0>
Traceback (most recent call last):
  File "/home/jeffery/anaconda3/envs/RL-server/lib/python3.10/site-packages/mujoco/egl/__init__.py", line 130, in __del__
  File "/home/jeffery/anaconda3/envs/RL-server/lib/python3.10/site-packages/mujoco/egl/__init__.py", line 123, in free
  File "src/errorchecker.pyx", line 58, in OpenGL_accelerate.errorchecker._ErrorChecker.glCheckError
OpenGL.raw.EGL._errors.EGLError: <exception str() failed>
```
wanna know if anybody has the solution for this problem.
Thx you !!

cidxb, 2023-03-13:
I  gonna close this topic , since I found https://github.com/deepmind/mujoco/issues/700 .

closed 2023-03-13 by cidxb

cidxb, 2023-03-14:
For fellows who will encounter glfw problem
here is a solution :
  - https://stackoverflow.com/questions/71010343/cannot-load-swrast-and-iris-drivers-in-fedora-35
  - export LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libstdc++.so.6
basically is the do a soft connection to your dll

saran-t, 2023-03-14:
@cidxb I'd be extremely cautious about doing that. Sideloading different versions of `libstdc++` is fairly likely to result in random, subtle errors affecting anything that's loaded into the process.

ctbfl, 2025-01-23:
Maybe because you do not use `env.close()` to close the env?