google-deepmind/mujoco #767: Plugin loading introduced in `3.2.0` can fail

milutter, 2023-03-14:
The loading of the plugins in `python/mujoco/__init__.py` can fail depending on the used build environment. When following the described installation process everything works fine. However, some build environments can "generously" add `__init__.py` files to all sub-directories, which causes the error `OSError: [...]/mujoco/site-packages/mujoco/plugin/__init__.py: invalid ELF header` as the added `plugin/__init__.py` file is passed to `ctypes.CDLL(...)`. 

While this failure is not caused by MuJoCo, it would be great to mitigate this failure. This failure could be prevented without much downsides by replacing [line 57](https://github.com/deepmind/mujoco/blob/2f0fb1e4ef81ae4eb281e09fb67c6b36f27dc427/python/mujoco/__init__.py#L57)
```
PLUGIN_HANDLES.append(ctypes.CDLL(os.path.join(directory, filename)))
```
 in `python/mujoco/__init__.py` with 
```
if os.path.splitext(filename)[-1] == ".so":
    PLUGIN_HANDLES.append(ctypes.CDLL(os.path.join(directory, filename)))
```
which ignores all files that are not a shared library. Alternatively one could also raise an explicit error when other files except a `__init__.py` is added, e.g., 
```
if os.path.splitext(filename)[-1] == ".so":
    PLUGIN_HANDLES.append(ctypes.CDLL(os.path.join(directory, filename)))

elif filename == "__init__.py":
    pass

else:
    raise ValueError(f"The file {filename} is not a shared library.")
```

What do you think?

saran-t, 2023-03-14:
@milutter Thanks! Would you mind putting this in a PR so that I can review it properly?

milutter, 2023-03-14:
@saran-t Sure. The PR is at https://github.com/deepmind/mujoco/pull/769

saran-t, 2023-03-16:
Fixed in #769

closed 2023-03-16 by saran-t