google-deepmind/mujoco #768: LengthRange Computation did not converge in actuator

Gokul2411, 2023-03-14:
I am student and am working on mujoco for muscle simulations.

I am currently working in extending arm26.xml model and am facing an error.

This is the xml file of the model

`<mujoco model="2-link 6-muscle arm">
  <option timestep="0.005" iterations="50" solver="Newton" tolerance="1e-10" gravity = "0 0 0"/>

  <visual>
    <rgba haze=".3 .3 .3 1"/>
  </visual>

  <default>
    <joint type="hinge" pos="0 0 0" axis="0 0 1" limited="true" range="0 120" damping="0.1"/>
    <muscle ctrllimited="true" ctrlrange="0 1"/>
  </default>

  <asset>
    <texture type="skybox" builtin="gradient" rgb1="0.6 0.6 0.6" rgb2="0 0 0" width="512" height="512"/>

    <texture name="texplane" type="2d" builtin="checker" rgb1=".25 .25 .25" rgb2=".3 .3 .3" width="512" height="512" mark="cross" markrgb=".8 .8 .8"/>

    <material name="matplane" reflectance="0.3" texture="texplane" texrepeat="1 1" texuniform="true"/>
  </asset>

  <worldbody>
    <geom name="floor" pos="0 0 -0.5" size="0 0 1" type="plane" material="matplane"/>

    <light directional="true" diffuse=".8 .8 .8" specular=".2 .2 .2" pos="0 0 5" dir="0 0 -1"/>

    <site name="Rs0" pos="0 0 0" size="0.02"/>
    <site name="Rx0" pos="0.15 -0.15 0" size="0.02" rgba="0 .7 0 1" group="1"/>

    <site name="Ls0" pos="0 0 0" size="0.02"/>
    <site name="Lx0" pos="-0.15 -0.15 0" size="0.02" rgba="0 .7 0 1" group="2"/>

    <body pos="0.15 0 0">
      <geom name="Rupper arm" type="capsule" size="0.045" fromto="0 0 0  0.5 0 0" rgba=".5 .1 .1 1"/>
      <joint name="Rshoulder"/>
      <geom name="Rshoulder" type="cylinder" pos="0 0 0" size=".1 .05" rgba=".5 .1 .8 .5" mass="0" group="1"/>

      <site name="Rs1" pos="0.15 0.06 0" size="0.02"/>
      <site name="Rs2" pos="0.15 -0.06 0" size="0.02"/>
      <site name="Rs3" pos="0.4 0.06 0" size="0.02"/>
      <site name="Rs4" pos="0.4 -0.06 0" size="0.02"/>
      <site name="Rs5" pos="0.25 0.1 0" size="0.02"/>
      <site name="Rs6" pos="0.25 -0.1 0" size="0.02"/>
      <site name="Rx1" pos="0.5 -0.15 0" size="0.02" rgba="0 .7 0 1" group="1"/>

      <body pos="0.5 0 0">
        <geom name="Rforearm" type="capsule" size="0.035" fromto="0 0 0  0.5 0 0" rgba=".5 .1 .1 1"/>
        <joint name="Relbow"/>
        <geom name="Relbow" type="cylinder" pos="0 0 0" size=".08 .05" rgba=".5 .1 .8 .5" mass="0" group="1"/>

        <site name="Rs7" pos="0.11 0.05 0" size="0.02"/>
        <site name="Rs8" pos="0.11 -0.05 0" size="0.02"/>
      </body>
    </body>

    <body pos="-0.15 0 0">
      <geom name="Lupper arm" type="capsule" size="0.045" fromto="0 0 0  -0.5 0 0" rgba=".5 .1 .1 1"/>
      <joint name="Lshoulder"/>
      <geom name="Lshoulder" type="cylinder" pos="0 0 0" size=".1 .05" rgba=".5 .1 .8 .5" mass="0" group="2" euler = "180 0 0"/>

      <site name="Ls1" pos="-0.15 0.06 0" size="0.02"/>
      <site name="Ls2" pos="-0.15 -0.06 0" size="0.02"/>
      <site name="Ls3" pos="-0.4 0.06 0" size="0.02"/>
      <site name="Ls4" pos="-0.4 -0.06 0" size="0.02"/>
      <site name="Ls5" pos="-0.25 0.1 0" size="0.02"/>
      <site name="Ls6" pos="-0.25 -0.1 0" size="0.02"/>
      <site name="Lx1" pos="-0.5 -0.15 0" size="0.02" rgba="0 .7 0 1" group="2"/>

      <body pos="-0.5 0 0">
        <geom name="Lforearm" type="capsule" size="0.035" fromto="0 0 0  -0.5 0 0" rgba=".5 .1 .1 1"/>
        <joint name="Lelbow"/>
        <geom name="Lelbow" type="cylinder" pos="0 0 0" size=".08 .05" rgba=".5 .1 .8 .5" mass="0" group="2" euler = "180 0 0"/>

        <site name="Ls7" pos="-0.11 0.05 0" size="0.02"/>
        <site name="Ls8" pos="-0.11 -0.05 0" size="0.02"/>
      </body>
    </body>
  </worldbody>

  <tendon>
    <spatial name="RSF" width="0.01">
      <site site="Rs0"/>
      <geom geom="Rshoulder"/>
      <site site="Rs1"/>
    </spatial>

    <spatial name="RSE" width="0.01">
      <site site="Rs0"/>
      <geom geom="Rshoulder" sidesite="Rx0"/>
      <site site="Rs2"/>
    </spatial>

    <spatial name="REF" width="0.01">
      <site site="Rs3"/>
      <geom geom="Relbow"/>
      <site site="Rs7"/>
    </spatial>

    <spatial name="REE" width="0.01">
      <site site="Rs4"/>
      <geom geom="Relbow" sidesite="Rx1"/>
      <site site="Rs8"/>
    </spatial>

    <spatial name="RBF" width="0.009" rgba=".4 .6 .4 1">
      <site site="Rs0"/>
      <geom geom="Rshoulder"/>
      <site site="Rs5"/>
      <geom geom="Relbow"/>
      <site site="Rs7"/> 
    </spatial>

    <spatial name="RBE" width="0.009" rgba=".4 .6 .4 1">
      <site site="Rs0"/>
      <geom geom="Rshoulder" sidesite="Rx0"/>
      <site site="Rs6"/>
      <geom geom="Relbow" sidesite="Rx1"/>
      <site site="Rs8"/>
    </spatial>

    <spatial name="LSF" width="0.01">
      <site site="Ls0"/>
      <geom geom="Lshoulder"/>
      <site site="Ls1"/>
    </spatial>

    <spatial name="LSE" width="0.01">
      <site site="Ls0"/>
      <geom geom="Lshoulder" sidesite="Lx0"/>
      <site site="Ls2"/>
    </spatial>

    <spatial name="LEF" width="0.01">
      <site site="Ls3"/>
      <geom geom="Lelbow"/>
      <site site="Ls7"/>
    </spatial>

    <spatial name="LEE" width="0.01">
      <site site="Ls4"/>
      <geom geom="Lelbow" sidesite="Lx1"/>
      <site site="Ls8"/>
    </spatial>

    <spatial name="LBF" width="0.009" rgba=".4 .6 .4 1">
      <site site="Ls0"/>
      <geom geom="Lshoulder"/>
      <site site="Ls5"/>
      <geom geom="Lelbow"/>
      <site site="Ls7"/> 
    </spatial>

    <spatial name="LBE" width="0.009" rgba=".4 .6 .4 1">
      <site site="Ls0"/>
      <geom geom="Lshoulder" sidesite="Lx0"/>
      <site site="Ls6"/>
      <geom geom="Lelbow" sidesite="Lx1"/>
      <site site="Ls8"/>
    </spatial>
  </tendon>

  <actuator>
    <muscle name="RSF" tendon="RSF"/>
    <muscle name="RSE" tendon="RSE"/>
    <muscle name="REF" tendon="REF"/>
    <muscle name="REE" tendon="REE"/>
    <muscle name="RBF" tendon="RBF"/>
    <muscle name="RBE" tendon="RBE"/>

    <muscle name="LSF" tendon="LSF" gear = 5/>
    <muscle name="LSE" tendon="LSE"/>
    <muscle name="LEF" tendon="LEF"/>
    <muscle name="LEE" tendon="LEE"/>
    <muscle name="LBF" tendon="LBF"/>
    <muscle name="LBE" tendon="LBE"/>
  </actuator>
</mujoco>`

The issue is with the actuator 6. It is throwing an error stating that LengthRange did not converge.

It is same as the Right hand of the model without any change in dimensions or connectivity. But still it is throwing an error.

Below here is the picture of the model.
[image]

I am able to actuate the right arm but when I actuate the left arm i am getting the below error.

[image]

What might be the issue here?

yuvaltassa, 2023-03-14:
Did you try setting different [parameter values](https://mujoco.readthedocs.io/en/latest/XMLreference.html#compiler-j-lengthrange)?

Gokul2411, 2023-03-14:
No, how to define tolrange in the ,xml file for a muscle actuator?

Gokul2411, 2023-03-14:
Thank you, after changing tolrange in complier/lengthrange it worked properly.

closed 2023-03-14 by Gokul2411