google-deepmind/mujoco #771: Simulating huge amount of soft bodies in MuJoCo

KuenMa, 2023-03-16:
Hello Mujoco team,

I'm an engineer who is using Mujoco for simulating a grasping task involving food.

My goal is to simulate a grasp and then measure how much food has been lifted.
I've successfully created a model that can simulate food that consists of around 50-100 ingredients. But increasing the amount, at some point I get the following error message:
`
WARNING:absl:Insufficient arena memory for the number of constraints generated. Increase arena memory allocation above 1904M bytes. Time = 0.0000`

So I've increased `<size memory` by 10 times and I get the same error: 
`WARNING:absl:Insufficient arena memory for the number of constraints generated. Increase arena memory allocation above 17179869182G bytes. Time = 0.0000.`

I'm at a loss here and would like some help with this memory allocation problem. Decreasing the number of ingredients is not preferable as it influences the grasping result.

I'v put a video in the appendix to show my setup (the food is moving at the beginning because I threw it down and didn't wait until it stopped)
The model consists of an area surrounded by walls where I put food inside. A gripper with which I want to grasp food. And food that I throw inside the specific area. In contrast to the walls and the gripper, the food can be very soft and bend, so I've decided to use composite to create soft ingredients. They can have different shapes like long and thin, flat or round. In the case of the video I'm using cloth to represent flat ingredients like Shougayaki (ginger pork).
One ingredient looks like this:

```
            <composite prefix="taira" type="cloth" count="8 3 1" spacing="0.0100" offset="0 0 0">
                <joint kind="main" damping="0.0002" stiffness="0.01"/>
                <joint kind="stretch" damping="0.009" stiffness="0.1"/>
                <geom group="1" type="capsule" size=".0075 .003" rgba=".8 .2 .1 1" mass="0.001"/>
            </composite>
```
If you need more information about my setup, please let me know.

https://user-images.githubusercontent.com/65333185/225486641-f207b505-df2f-4ccc-95d5-ac425c0a39e6.mp4

KuenMa, 2023-03-22:
@nimrod-gileadi 
Huge thank you for reaching out!
I'll definitely try the new version and see how far I can get with the new improvements and changes. I will get back when I have some new results to share~

nimrod-gileadi, 2023-04-20:
Feel free to reopen if there are further questions.

closed 2023-04-20 by nimrod-gileadi