google-deepmind/mujoco #772: Integraing ImGui with MuJoCo

Sam-0505, 2023-03-16:
**How to ask for help**

First, read our quick guide to
[asking good questions](https://github.com/deepmind/mujoco#asking-questions).
Below is a template for you to use:

Hi,

I'm a research intern and I'm trying to use MuJoCo for a RL research project.

I'm looking for some help with integrating Dear ImGui to the project.

Here is a part of the code where it renders the ouput:
```
while (!glfwWindowShouldClose(window)) {
        mjtNum simstart = d->time;
        while (d->time - simstart < 1.0 / 60.0) {
            mj_step(m, d);
        }

        ImGui_ImplOpenGL3_NewFrame();
        ImGui_ImplGlfw_NewFrame();
        ImGui::NewFrame();

        // get framebuffer viewport
        mjrRect viewport = { 10, 10, 0, 0 };
        glfwGetFramebufferSize(window, &viewport.width, &viewport.height);

        ////render your GUI
        ImGui::Begin("Demo window");
        ImGui::Button("Hello!");
        ImGui::End();

        ImGui::Render();
        ImGui_ImplOpenGL3_RenderDrawData(ImGui::GetDrawData());

        if (io.ConfigFlags & ImGuiConfigFlags_ViewportsEnable)
        {
            ImGui::UpdatePlatformWindows();
            ImGui::RenderPlatformWindowsDefault();
        }
        
        // update scene and render
        mjv_updateScene(m, d, &opt, NULL, &cam, mjCAT_ALL, &scn);
        mjr_render(viewport, &scn, &con);

        // swap OpenGL buffers (blocking call due to v-sync)
        glfwSwapBuffers(window);

        // process pending GUI events, call GLFW callbacks
        glfwPollEvents();
    }
```
</details>

Here is a screenshot / video, illustrating my question:

https://user-images.githubusercontent.com/65815740/225508372-ef2dc62d-8669-4d7e-bed2-cb268535956f.mp4

rohit-kumar-j, 2023-03-28:
I had created the same thing previously😅, however, the mouse events were being forwarded to the scene even when the mouse was not hovering (and was focused) on the imgui widget, since there was no input event abstarction layer.

@saran-t, I was hoping to find a solution of create a texture from the `mjvScene` like so:

```C++
uint32_t textureID = m_Framebuffer->GetColorAttachmentRendererID();
ImGui::Image((void*)textureID, 
            ImVec2{ m_ViewportSize.x, m_ViewportSize.y }, 
            ImVec2{ 0, 1 },  // optional: remapping the coordinate frame 
            ImVec2{ 1, 0 }); // optional: remapping the coordinate frame
```
and then pass it to imgui. This way I can handle all events inside imgui's viewport. Is there a native method for this in mujoco or should we set this up externally?

[the above code is from here](https://github.com/TheCherno/Hazel/blob/ffb53e932e2ddbf3112aca31bef291295ecf4906/Hazelnut/src/EditorLayer.cpp#L167)

~I think that this method may be slow if it is multithreaded, as the every time a fetch call is made to the imgui state, the cpu cache is flused. Perhaps only do this for debugging~

gedalia, 2023-12-10:
ImGui::Render(); Really should be the last thing you do before swapping buffers. It’s a UI this that’s going over the 3D scene,

kevinzakka, 2024-01-03:
Marking as fixed. Feel free to re-open if you haven't solved your issue with the above comments.

closed 2024-01-03 by kevinzakka

gedalia, 2024-01-06:
BTW, slightly off-topic but one thing that might improve debugging here is if mujoco rendering used more modern OpenGL calls, currently NSight can't really step through mujoco rendering due to it's use of the fixed function pipeline and legacy ogl calls. I'm wondering if there are any plans to modernize that?