google-deepmind/mujoco #773: How to set a margin for the contact

rongakowang, 2023-03-17:
For the setting below:

[image]

The thumb is not actually contacting the object, but I wish the simulator to tolerate this and include this and include a contact force from the thumb to the object, as if there is a margin outside the thumb, what should I do for this? I have tried to set a positive `margin` outside the `geom` of the object mesh but it has no effects.

yuvaltassa, 2023-03-17:
`<geom margin/>` should work.

If you attach a loadable model we can help you debug.

rongakowang, 2023-03-17:
Hi @yuvaltassa :

Thank you for your assistance! I have attached the model to reproduce the scene:

[debug file.zip](https://github.com/deepmind/mujoco/files/11000334/debug.file.zip)

I wish the hand to stablely grasp the object, however, I am also not sure how to achieve it because the object will just slip.

Please note the hand is set to be static and the object is free

yuvaltassa, 2023-03-17:
I played with this. I don't think your hand can grasp the object. This is not _grasping_, this is _jamming_.

Without giving you hand degrees of freedom, any sort of tiny slip will lead to more slip. There is no way to get a good grasp with a static hand. Perhaps you should try using [adhesion actuators](https://mujoco.readthedocs.io/en/latest/XMLreference.html#actuator-j-adhesion)?

closed 2023-03-17 by yuvaltassa