google-deepmind/mujoco #774: Commit 70959c1 breaks the simulate viewer

aftersomemath, 2023-03-17:
Commit 70959c1a2ccf4169b30e31eec596c183e167d8c3 causes the simulate viewer (C++ and Python) to crash upon loading an XML file. For example:

```
$ bin/simulate ../model/humanoid100/humanoid100.xml
MuJoCo version 2.3.2
ERROR: mjui_add: data pointer for item without data

Press Enter to exit ...
```

The Python version of simulate is broken too. Example:

```
$ python -m mujoco.viewer --mjcf=model/humanoid100/humanoid100.xml
ERROR: mjui_add: edit length missing

Press Enter to exit ...
```

I started with the tip of `main` and bisected to find that 47b58f5ccb9c1b1fe3c42f4d9b8727978c45b82a works but the next commit, 70959c1a2ccf4169b30e31eec596c183e167d8c3, does not. I tried a clean build but that did not help.

Context:
- Ubuntu 22.04
- gcc 11.3

nimrod-gileadi, 2023-03-17:
Are you sure you don't have local edits?

I just tried building from `main` on a linux machine (using Clang 15, not gcc 11) and running the same `simulate` command, and I don't get the crash.

saran-t, 2023-03-17:
@aftersomemath I cannot reproduce this locally. Everything is working fine for me on macOS.

aftersomemath, 2023-03-17:
I just tried Clang 14 and `simulate` worked. So the problem appears to be at the least specific to gcc.

I triple checked for that there are no local edits.

nimrod-gileadi, 2023-03-17:
I found the issue. Fix will come soon.

closed 2023-03-17 by copybara-service[bot]

nimrod-gileadi, 2023-03-20:
Thanks for catching this!

aftersomemath, 2023-03-20:
You're welcome!