google-deepmind/mujoco #776: Memory misalignment with output of mj_makedata(m)

bbokser, 2023-03-19:
Hi,

I have been getting a strange issue running my code on a new computer. Calling variables from the `d` struct (the output of mj_makedata(m) results in a segfault:

```
Stack trace (most recent call last):
#6    Object "[0xffffffffffffffff]", at 0xffffffffffffffff, in 
#5    Object "/home/ben-linux/git-workspace/RosDockerWorkspace/devel/lib/hopper_mpc/hopper_mpc", at 0x55c60c61786d, in _start
#4    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7fa6df1a4082, in __libc_start_main
#3    Object "/home/ben-linux/git-workspace/RosDockerWorkspace/devel/lib/hopper_mpc/hopper_mpc", at 0x55c60c61a1b0, in main
#2    Object "/home/ben-linux/git-workspace/RosDockerWorkspace/devel/lib/hopper_mpc/hopper_mpc", at 0x55c60c637e5b, in Runner::Run()
#1    Object "/home/ben-linux/git-workspace/RosDockerWorkspace/devel/lib/hopper_mpc/hopper_mpc", at 0x55c60c666d54, in MujocoBridge::Init(double)
#0    Object "/home/ben-linux/git-workspace/RosDockerWorkspace/devel/lib/hopper_mpc/hopper_mpc", at 0x55c60c6652d9, in mujoco_set_qpos
Segmentation fault (Address not mapped to object [(nil)])
[1]    34369 segmentation fault  rosrun hopper_mpc hopper_mpc mujoco start_stand stand 5000 --plot
```

It appears variables such as `qpos` and `qvel` are ceasing to exist (pardon my lack of vocabulary) once mj_makedata(m) is written to the `mjData* d` variable, although they briefly exist *within* the mj_makedata() function. 

I know this because I built MuJoCo with debugging enabled and stepped through the code line by line until the moment that `d` becomes corrupted. I also modified the source code to print the size of `d`, in order to demonstrate this:

```
mjData *mj_makeData(const mjModel *m) {
  mjData *d = _makeData(m);
  if (d) {
    mj_resetData(m, d);
  }
  printf("size of d in mujoco  = %lu\n", sizeof(*d));
  return d;
}
```
And inside my code:

```
printf("size of d = %lu\n", sizeof(*d));
// make data
d = mj_makeData(m);
```

The result in the terminal is this:
```
size of d in mujoco  = 41240
size of d = 41160
size of d in mujoco  = 41240
```

Because my code already demonstrably worked on a different computer, I suspect there is a difference in the compilers used (I am using ROS noetic catkin, and I tried updating cmake to the latest version--didn't help), and issue is being caused by the fact that I am writing in C++ whereas MuJoCo is written in C. Because of this, I tried heavily modifying my code to use a C wrapper (see [here](https://github.com/RoboticExplorationLab/RExHopper/blob/bugfix/mujoco/hopper_mpc/src/bridge_mujoco_wrapper.c)) in order to avoid exposing `d` to C++, but this did not fix the issue. So this is beyond my understanding.

I also suspected that the difference is being caused by the use of a newer version of MuJoCo, so I tried reverting to the 2.2.0 version, which I believe is what was installed on the other computer. This did not help.

Finally, I should note that the MuJoCo examples, such as `basic.cc`, still work, even if I add calls to `d->qpos`.

I am at a loss as to what to do here.

DavidPL1, 2023-03-20:
I am building MuJoCo 2.3.2 form source on focal and using it in a C++ wrapping library (with mjData and mjModel exposed to C++) with ROS Noetic and I'm not experiencing any issues. 

Are you building MuJoCo with Catkin? 
I'm either building it externally with plain cmake and installing it to the devel/install space of my catkin workspace, or using an environment variable that points to an extracted release and just use a custom [Findmujoco.cmake](https://github.com/ubi-agni/mujoco_ros_pkgs/blob/noetic-devel/mujoco_ros/Findmujoco.cmake) file that is able to handle both cases.

saran-t, 2023-03-20:
Please post the segfaulting code.

bbokser, 2023-03-20:
Hi,

I'm building MuJoCo with cmake:
```
cd ~ && git clone https://github.com/deepmind/mujoco.git \
    && cd ~/mujoco && mkdir build_dir && cd build_dir \
    && sudo cmake .. -DCMAKE_INSTALL_PREFIX=/opt/mujoco \
    && sudo cmake --build . && sudo make install
```

Here is how I am setting up my environment and installing MuJoCo:
https://github.com/RoboticExplorationLab/RExHopper/blob/devel/docs/EnvSetup.md

And here is the segfaulting code.
The C wrapper version:
https://github.com/RoboticExplorationLab/RExHopper/blob/bugfix/mujoco/hopper_mpc/src/bridge_mujoco_wrapper.c
(segfaults at line 154 when `d->qpos[0]` is called inside `mujoco_set_qpos()` called at line 83 [here](https://github.com/RoboticExplorationLab/RExHopper/blob/bugfix/mujoco/hopper_mpc/src/bridge_mujoco.cpp))

The wrapperless C++ version:
https://github.com/RoboticExplorationLab/RExHopper/blob/devel/hopper_mpc/src/bridge_mujoco.cpp
(segfaults at line 210 when `d_global->qpos[0]` is called)

saran-t, 2023-03-21:
It's very difficult to be certain about what your code is actually doing without spending a lot of time absorbing the whole codebase in detail.

Why are you using global variables at all? Why aren't the MuJoCo data structures members of the classes that require them?

nimrod-gileadi, 2023-03-21:
Could it be that the mjdata.h header file you're using to build your own code is different from the header file used to build MuJoCo itself? That would explain the difference in sizeof.

Maybe if you have a dirty build or install directory? Or for some reason you're picking up the includes for MuJoCo from the wrong directory?

bbokser, 2023-03-22:
> Could it be that the mjdata.h header file you're using to build your own code is different from the header file used to build MuJoCo itself? 

That fixed it! It was the mjdata.h header file. I must have been testing my code against the wrong version of MuJoCo even when I reverted to the 2.2.0 version. I should not be linking MuJoCo this way.

> Why are you using global variables at all? Why aren't the MuJoCo data structures members of the classes that require them?

This is also an extremely valid criticism--most likely it was due to my incompetence with C++, although I don't remember the exact issue I encountered when I originally wrote the code.

closed 2023-03-22 by bbokser