google-deepmind/mujoco #777: Modelling adhesion between two static meshes

rongakowang, 2023-03-20:
Thank you for providing the [sample](https://github.com/deepmind/mujoco/blob/main/model/adhesion/active_adhesion.xml) for modelling an adhesion actuator, however, I wonder if it's possible to model the adhesion between a free object and a static mesh (without any degree of freedom), as per #773 ?

Could you please kindly provide a minimal xml for modelling this? Thanks in advance.

rongakowang, 2023-03-20:
Hi @yuvaltassa :

Thank you for your reply, I have tried adding adhesion directly on the body but it seems having no effect:

[debug file.zip](https://github.com/deepmind/mujoco/files/11023897/debug.file.zip)

Basically I wish the object to be attached, not falled, as long as it has contact with the mesh

Also I wonder if it's possible to set a threshold for the adhesion, i.e. only to stablely adhere to the mesh if there is sufficiently enough contact points.

yuvaltassa, 2023-03-21:
The model you sent doesn't have an adhesion actuator?

I added this and it works (obviously you need to set the ctrl to some nonzero value)
```xml
<actuator>
  <adhesion name="hand" body="hand" ctrlrange="0 1" gain="100"/>
</actuator>
```