google-deepmind/mujoco #781: Question about the rendering resource consume in graphic card

cidxb, 2023-03-21:
Hey.
 
**Background Info**
I use the mujoco +  Stable-Baselines3 - ( Reinforcement Learning Implementations)
To train agent with multi-processing (using CPU core to open Mujoco threading parallelly as much as I need)

**Question about the rendering memory cast in graphic card** 
But I find a strange thing, the same script, the same procedure in two computers.

During the simulation, one shows 
[image]

The other one shows 
[image]

The first image indicates that I have so many mujoco threading running, and the rendering of them is consuming graphic card resources, which makes sense.

But the second image of another computer shows only one main threading, and it is doing calculation "C", not graphic rendering "G".

I try to set the environment variable `MUJOCO_Gl` into the same `MUJOCO_GL=GLFW`, and the consuming is the same.

Because the SB3 is just parallel opening MUJOCO, so I think I must do something different in setting up mujoco, but I don't know what I have done wrong, can some friends from the community point it out to me ?
THX !

saran-t, 2023-03-21:
You haven't provided us with enough information to be able to tell what might be happening here. What's the script you're trying to run? What's the system configuration? How's your Python environment set up?

cidxb, 2023-03-22:
> You haven't provided us with enough information to be able to tell what might be happening here. What's the script you're trying to run? What's the system configuration? How's your Python environment set up?

THX your replay! I have edited my original post, and added as much info as I can .

saran-t, 2023-03-23:
Not really sure what's going on, but my suggested first thing to try is `MUJOCO_GL=EGL` instead of GLFW.

cidxb, 2023-03-24:
> Not really sure what's going on, but my suggested first thing to try is `MUJOCO_GL=EGL` instead of GLFW.

I don't know why, but if I used `MUJOCO_GL=EGL`

There will be problems like this : 
`Exception ignored in: <function GLContext.__del__ at 0x7f1c14000dc0>
Traceback (most recent call last):
  File "/home/bjae/anaconda3/envs/RL-server/lib/python3.10/site-packages/mujoco/egl/__init__.py", line 130, in __del__
  File "/home/bjae/anaconda3/envs/RL-server/lib/python3.10/site-packages/mujoco/egl/__init__.py", line 123, in free
  File "src/errorchecker.pyx", line 58, in OpenGL_accelerate.errorchecker._ErrorChecker.glCheckError
OpenGL.raw.EGL._errors.EGLError: <exception str() failed>`

I checked with this error alert script in errorchecker, still have no clue. 

and 

`X Error of failed request:  BadAccess (attempt to access private resource denied)
  Major opcode of failed request:  152 (GLX)
  Minor opcode of failed request:  5 (X_GLXMakeCurrent)
  Serial number of failed request:  191
  Current serial number in output stream:  191`

For this error, I also tried to fix it, from the information I gather, it should be an X server error that normally occurred in dueling with a remote SSH server is something with memory resources. BUt strange thing is, I get this problem in local host. And I have tried some solutions I found online. That's why I was using GLFW.

saran-t, 2023-03-30:
It's very difficult for me to help since whatever is happening seems to be specific to your setup. Is there Docker involved in the run that's not utilising GPU?

cidxb, 2023-03-31:
> 

Thank you for your reply! I solved this problem, it is kind of a dumb reason : 
This is because the Lenovo laptop runs a dynamic graphic setting that will automatically change the graphic for rendering. The GTX is used for calculation because in the script it chose a cuda device to do that, but for the rendering, the i915 intel graph card has taken over.

So I go to bios and set the dynamic graphic setting to Discrete Graphic, and the problem solved.

And the following command helped me debug 👍, I would to share:

`  $ gpustat`  helps me realize the GPU usage of GTX is 0%
` $  lspci -k | grep -A 2 -i "VGA"`   helps me to see how many graphic cards in the device
`  $ nvidia-setting` helps me realize the GTX GPU current is not in used because it does not show the window of X server setting, and it is almost empty.
  `$ sudo intel_gpu_top`  helps me confirm I am running the rendering of mujoco on the core graphic card

closed 2023-03-31 by cidxb

simeon-ned, 2024-10-08:
> > It's very difficult for me to help since whatever is happening seems to be specific to your setup. Is there Docker involved in the run that's not utilising GPU?
> 
> Thank you for your reply! I solved this problem, it is kind of a dumb reason : This is because the Lenovo laptop runs a dynamic graphic setting that will automatically change the graphic for rendering. The GTX is used for calculation because in the script it chose a cuda device to do that, but for the rendering, the i915 intel graphic card has taken over.
> 
> So I go to bios and set the Dynamic Graphic setting to Discrete Graphic, and the problem was solved.( The original idea is to disable the security Bios boot according to the post I saw online, but instead, I found the dynamic graphic setting.)
> 
> And the following command helped me debug 👍, I would like to share:
> 
> ` $ gpustat` helps me realize the GPU usage of GTX is 0% ` $ lspci -k | grep -A 2 -i "VGA"` helps me to see how many graphic cards in the device ` $ nvidia-setting` helps me realize the GTX GPU current is not in used because it does not show the window of X server setting, and it is almost empty. `$ sudo intel_gpu_top` helps me confirm I am running the rendering of mujoco on the core graphic card

The setting to discrete graphics indeed helps, thanks!