google-deepmind/mujoco #782: Python viewer (v 2.3.3)

vikashplus, 2023-03-22:
Running multiple viewer (even with same model and data) results in a runtime error
```
from mujoco import viewer
    viewer.launch_passive(model, data)
    viewer.launch_passive(model, data)

> _MJPYTHON.launch_on_ui_thread(model, data)
> File "<string>", line 28, in launch_on_ui_thread
> RuntimeError: another MuJoCo viewer is already open
```
What should be the intended behavior here? 
- Get two viewers?
- Just return back the original viewer?

Is it possible for the viewer launch call to return a handle of some kind?
`handle = viewer.launch_passive(model, data)`

saran-t, 2023-03-22:
Multiple viewers is currently not supported on macOS. Are you making a feature request? :)

saran-t, 2023-03-23:
A handle into the viewer is slated for next release. The proposed API is a context manager.

```python
with viewer.launch_passive(m, d) as v:
  # run whatever code you want
```

That way you have explicit control over when the viewer window should close, however it _does_ require that user code goes inside the context.

Does that sound good to you?

vikashplus, 2023-03-23:
I'm not sure if this will be ideal. 

For simple scripts this make sense to me. Usually viewer is a member of the task/env-class. These classes are constructed once and tossed around the RL agents for training and evaluations loops.

saran-t, 2023-04-10:
@vikashplus I'd strongly advise against putting the viewer inside a task or env class. Viewers should take a running env and visualise it. Environments themselves shouldn't own the visualiser...

vikashplus, 2023-04-12:
I see your point. I can to see how we can refactor things. 

Any recommendation on how to get the viewer-options so that we can configure things? The primary thing I want to be able to do is to control the position of the default camera, and toggle a few flags.