google-deepmind/mujoco #783: stack overflow (v2.3.3)

vikashplus, 2023-03-22:
Its not deterministically reproducible but I did observe the following stack overflow. IIRC that we changed something in the memory footprint, maybe something related to it?
```
/python3.7/site-packages/mujoco/viewer.py", line 104, in _reload
    mujoco.mj_forward(m, d)
mujoco.FatalError: stack overflow: max = 56623104, available = 56600292, requested = 19761945272 (ne = 5, nf = 6, nefc = 21, ncon = 0)
```

vikashplus, 2023-03-22:
Let me try to catch it.  Next time it happens I'll try to zero in on these details.

saran-t, 2023-04-10:
I think I've identified the responsible race condition (same thing as in https://github.com/deepmind/mujoco/issues/790).

Fix incoming.

commit 34226bf5 (2023-04-10): Fix a race condition in `launch_passive` and `launch_repl`.

Before this change, these functions can return before an internal call to `mj_forward`, which results in a race condition if the user script then tries to modify the physics state.

Fixes #783
Fixes #790

PiperOrigin-RevId: 523214378
Change-Id: I8a4c6cef0f5d3ec781fd2c5cae3c20dd4b8be67f
closed 2023-04-10 by copybara-service[bot]