google-deepmind/mujoco #784: Adding custom PyOpenGL code to MuJoCo context

SridharPandian, 2023-03-22:
I have a custom dm_control environment which renders 6 camera faces (for generating a cubemap) headlessly through EGL. I am trying to add some OpenGL v3.3 (using PyOpenGL) code that contains shaders which transform the cubemap to a fisheye image. On running both together, I found that the custom OpenGL code could not find a context for rendering. It gives the following error:
`OpenGL.error.Error: Attempt to retrieve context when no valid context`

The OpenGL code works when I run it separately. Is there a way to add the custom OpenGL code to the MuJoCo context? Also, the custom OpenGL code contains a new framebuffer, textures objects and vertex buffers. Would this cause issues when I add them in the same context?

Thanks!

SridharPandian, 2023-03-23:
Attaching my code here for reference.

SridharPandian, 2023-03-23:
Attaching code for reference:
`# Initialiizing the environment
env = control.Environment(
    physics, task, time_limit=300, control_timestep=control_timestep,
    **environment_kwargs)

# Object containing the PyOpenGL code
renderer = CubeMapRenderer()

spec = env.action_spec()
random_state = np.random.RandomState(42)

while env.physics.data.time < 7:
    action = random_state.uniform(spec.minimum, spec.maximum, spec.shape)
    time_step = env.step(action)    
    obs = time_step.observation['rgb']
    images = [obs[key] for key in obs.keys()]
    fisheye = renderer.render_fisheye_image(images) # Breaks here
`
The `CubeMapRenderer` class looks like this:
`
class CubeMapRenderer:
    def __init__(self, output_image_size = (1024, 1024)):
        self.output_image_size = output_image_size

        self._load_vertex_buffers()
        self._create_framebuffer()
        self._load_shaders()

    def _load_shaders(self):
        # vertex_shader_path = SHADER_PATH / 'vertex_shader'
        # fragment_shader_path = SHADER_PATH / 'fragment_shader'

        vertex_shader_path = SHADER_PATH / 'kb_vertex_shader'
        fragment_shader_path = SHADER_PATH / 'kb_fragment_shader'
        
        with open(vertex_shader_path, 'rb') as shader_file:
            vertex_shader_src = shader_file.read()
        
        with open(fragment_shader_path, 'rb') as shader_file:
            fragment_shader_src = shader_file.read()

        self.vertex_shader = shaders.compileShader(vertex_shader_src, GL_VERTEX_SHADER)
        self.fragment_shader = shaders.compileShader(fragment_shader_src, GL_FRAGMENT_SHADER)

        self.shaders = shaders.compileProgram(
            self.vertex_shader,
            self.fragment_shader
        )

        glUseProgram(self.shaders)

    def _load_vertex_buffers(self):
        # Creating the vertex array object
        self.vao = glGenVertexArrays(1)
        glBindVertexArray(self.vao)

        # Creating the vertex buffer object
        self.vbo = glGenBuffers(1)
        glBindBuffer(GL_ARRAY_BUFFER, self.vbo)
        glBufferData(GL_ARRAY_BUFFER, VERTICES.nbytes, VERTICES, GL_STATIC_DRAW)

        # Assigning the vertex attribute pointers
        stride = VERTICES.itemsize * 5
        glVertexAttribPointer(0, 3, GL_FLOAT, GL_FALSE, stride, ctypes.c_void_p(0))
        glEnableVertexAttribArray(0)
        glVertexAttribPointer(1, 2, GL_FLOAT, GL_FALSE, stride, ctypes.c_void_p(12))
        glEnableVertexAttribArray(1)

    def _create_framebuffer(self):
[... 62 lines cut ...]
        glBindTexture(GL_TEXTURE_CUBE_MAP, 0)
        glDeleteTextures([cubemap_texture])

        return np.flip(image_data, axis = 0)

    def close(self):
        # Detaching and deleting the shaders
        glUseProgram(0)
        glDetachShader(self.shaders, self.vertex_shader)
        glDetachShader(self.shaders, self.fragment_shader)
        glDeleteProgram(self.shaders)

        # Unbinding and deleting the vertex buffers
        glBindVertexArray(0)
        glBindBuffer(GL_ARRAY_BUFFER, 0)
        glDeleteVertexArrays(1, [self.vao])
        glDeleteBuffers(1, [self.vbo])

        # Unbinding textures
        glBindTexture(GL_TEXTURE_2D, 0)
        glDeleteTextures([self.output_texture])

        # Unbinding the framebuffer
        glBindFramebuffer(GL_FRAMEBUFFER, 0)
`

saran-t, 2023-03-30:
The problem here is that MuJoCo's renderer is "old school OpenGL", i.e. fixed pipeline, pre-shader OpenGL. This means that its context doesn't support OpenGL 3 (or anything that uses shaders, for that matter).

You may be able to create your own context for your shader, but then you'll have to figure out how to best get images from MuJoCo's renderer into textures for your own renderer.

I'm sorry that it's probably not the answer you're expecting! I'm not the biggest fan of the existing renderer code either, but it's just not high enough priority for us to rewrite the whole renderer at this time.

closed 2023-03-30 by saran-t