google-deepmind/mujoco #785: How to update `bvh_aabb` after changing the com of a body in the model at runtime?

breakds, 2023-03-23:
Hi,

This is a question related to `bvh_aabb` which is introduced in this commit https://github.com/deepmind/mujoco/commit/70959c1a2ccf4169b30e31eec596c183e167d8c3 

In order to perform domain randomization of object properties such as `mass` and `com` of `<body>` without having to re-parse and re-load the `xml` MJCF definitions, we chose to directly update the model as described in #764.

After updating the `com` or `mass` of `<body>`, the model becomes inconsistent, but after a call to `mj_setConst`, the consistency is restored. In the latest version of `mujoco`, I found that the field `bvh_aabb` will not be updated after `mj_setConst`. My questions are:

1. In this case, if model has a inconsistent `bvh_aabb` but I still continue the simulation, what will be the impact?
2. Is there a way to re-compute `bvh_aab` after updating the model, so that I can restore the consistency of the model?

Thanks a lot!

breakds, 2023-03-23:
The problem is that re-compiling the XML is very slow, especially when we are only modifying one single property for the domain randomization purpose in RL training. Is there a suggested way to do this faster?

erez-tom, 2023-03-23:
Do you know why compilation is so slow for you? Are you loading large assets (e.g., meshes) from disk?

yuvaltassa, 2023-03-23:
To add: slow compilation is something we are actively working on and intend to address this year. While it is possible to safely change some values in mjModel, this is not the correct long term approach, and we try to discourage it.

That said, as Tom implies, there are various ways to speed up compilation now.

breakds, 2023-03-23:
Thanks for the reply! 

@erez-tom  Our scene has meshes for the robots (which are usually a few megabytes and in total < 10 megabytes), and a height field (1600 x 1600 vertices), which is 3.5 megabytes. In this case, is there a way to recompile the XML file without having to reload the assets?

Thank you!