google-deepmind/mujoco #787: How to generate references.h for documentation

DavidPL1, 2023-03-23:
I really like the style of the documentation with sphinx and was looking to build documentation for my own MuJoCo extension the same way by shamelessly copying most of your sphinx config.

The [doc/includes/references.h](https://github.com/deepmind/mujoco/blob/main/doc/includes/references.h) file mentions that it is generated automatically by scanning the header files, but I could not find code generating that file.

Could you provide some insight on how it's done?

nimrod-gileadi, 2023-04-20:
Our parsing is quite specific to how MuJoCo header files are written, and probably won't help your project.

You can take a look at https://github.com/deepmind/mujoco/blob/main/doc/ext/header_reader.py

We have a wrapper script (at the moment, not open source because it's tied to our internal build setup), which header_reader.read on every relevant header file, then formats and concatenates the result into references.h.

closed 2023-04-20 by nimrod-gileadi

DavidPL1, 2023-04-20:
I'll have a look, thank you!

saran-t, 2023-04-20:
Another possibility is to use Clang to parse your source file. The nice thing about this approach is since you're using a compiler frontend, it will actually deal with things like includes and typedefs for you, and generate a JSON file full of information about your code. It's also going to be less likely to break if you reformat your code.

The command to use is e.g. [as given here in `introspect`](https://github.com/deepmind/mujoco/blob/main/introspect/codegen/generate_functions.py#L18). You'd then need to write code to transform the JSON into whatever documentation you need.