google-deepmind/mujoco #789: Mocap issue when welding the mocap body and robot gripper

Rancho-zhao, 2023-03-24:
Hi, I'm a student and I'm trying to use MuJoCo for robot manipulation tasks.

I want to use mocap to control my robot gripper to grasp objects. But when I weld the mocap body and the gripper, something unexpected happend, that the relative pose between the mocap body and the gripper is not as my set.

Here is a model which explains my question:

<details>
  <summary>minimal XML</summary>

```XML
<?xml version="1.0" encoding="utf-8"?>
<mujoco>
	<compiler angle="radian"  coordinate="local" meshdir="../stls/robot_tactile" ></compiler>
	<option timestep="0.002">
		<flag warmstart="enable" multiccd="enable"></flag>
	</option>
    <default>
        <light castshadow="false" diffuse="1 1 1" />
        <camera fovy="60" />
        <motor ctrllimited="true" ctrlrange="-0.04 0.04"/>
        <!-- <joint range="-1.5708 1.5708"/> -->
    </default>

    <visual>
        <!-- <rgba connect="1 0.2 1 1" joint="0.2 0.6 0.8 1"/> -->
        <map fogstart="3" fogend="5"/>
        <quality shadowsize="2048"/>
    </visual>
    <option gravity="0 0 -9.81" iterations="100"/>

    <custom>
        <numeric name="N_GRIPPER_JOINTS" data="0" />
    </custom>

    <equality>
        <weld body1="robot0:mocap" body2="gripper_base"></weld> 
    </equality>
    
    <size nconmax="2000" njmax="10000" nstack="500000"/>

	<include file="shared.xml"></include>
	
	<worldbody>

		<geom name="floor0" pos="0 0 0" size="0.85 0.7 1" type="plane" contype="32" conaffinity="32" condim="3" material="floor_mat"></geom>
		<body name="floor0" pos="0 0 0">
			<site name="target0" pos="0 0 0.5" size="0.02 0.02 0.02" rgba="1 0 0 0.0" type="sphere"></site>
		</body>
        <body mocap="true" name="robot0:mocap" pos="0 0.092 0.9">
            <geom conaffinity="0" contype="0" size="0.02 0.02 0.02" type="box"></geom>
        </body>

        <include file="gripper.xml"></include>
        <!-- box for the obect to sit on -->
        <body name="table0" pos="-.2 0 .6">
            <body name="object_table" pos="0 0 0">
                <!-- surface -->
                <geom name="surface" type="box" pos="0 0 0" size=".275 .275 .015" conaffinity="32" contype="32" mass="1" rgba="1 1 1 1" friction="1 0.05 0.01" solimp="1.1 1.2 0.001 0.5 2" solref="0.02 1"/>
                
                <!-- valid spawn areas for target objects and positions -->
                <!-- makes the objects spawn in front of the arm only-->
                <geom name="spawn surface" type="box" pos="-.095 0 .016" size=".12 .245 .00001" conaffinity="0" contype="0" mass="0" rgba="1 1 0 0"/>

                <!-- legs -->
                <body pos="0 0 -.3">
                    <geom name="leg0" type="box" pos="0.26 0.26 0" size="0.015 0.015 .3" conaffinity="32" contype="1" friction="1 0.05 0.01" solimp="1.1 1.2 0.001 0.5 2" solref="0.02 1"/>
                    <geom name="leg1" type="box" pos="-.26 -.26 0" size="0.015 0.015 .3" conaffinity="32" contype="1" friction="1 0.05 0.01" solimp="1.1 1.2 0.001 0.5 2" solref="0.02 1"/>
                    <geom name="leg2" type="box" pos="0.26 -.26 0" size="0.015 0.015 .3" conaffinity="32" contype="1" friction="1 0.05 0.01" solimp="1.1 1.2 0.001 0.5 2" solref="0.02 1"/>
                    <geom name="leg3" type="box" pos="-.26 0.26 0" size="0.015 0.015 .3" conaffinity="32" contype="1" friction="1 0.05 0.01" solimp="1.1 1.2 0.001 0.5 2" solref="0.02 1"/>
                </body>

            </body>
        </body>  

        <body name="object0" pos="-0.215 0.1 .66" euler="0 0 0">
            <site name="object0:site" pos="0 0 0" rgba="0 1 0 .1" size="0.02 0.02 0.02"/>
            <joint name="object0:joint" type="free"/>
            <geom name="object0" type="box" size=".02 .02 .02" mass=".001" contype="1" conaffinity="32" condim="3" rgba="1 1 0 1" />
        </body>

        <body name="object1" pos="-0.225 0.1 .62">
            <site name="object1:site" pos="0 0 0.04" rgba="0 0 0 0" size=".02 .02 .02"/>
            <joint name="object1:slide1" pos="0 0 0" type="slide" axis="1 0 0" damping="1e+11"/>
            <joint name="object1:slide2" pos="0 0 0" type="slide" axis="0 1 0" damping="1e+11"/>
            <geom name="left" type="box" pos="-.041 0 .02" size=".02 .06 .02" mass="1" contype="32" conaffinity="32" condim="4" rgba=".5 .5 .5 1"/>
            <geom name="right" type="box" pos=".041 0 .02" size=".02 .06 .02" mass="1" contype="32" conaffinity="32" condim="4" rgba=".5 .5 .5 1"/>
            <geom name="front" type="box" pos="0 .041 .02" size=".06 .02 .02" mass="1" contype="32" conaffinity="32" condim="4" rgba=".5 .5 .5 1"/>            
            <geom name="back" type="box" pos="0 -.041 .02" size=".06 .02 .02" mass="1" contype="32" conaffinity="32" condim="4" rgba=".5 .5 .5 1"/>
        </body>

        <light directional="true" ambient="0.2 0.2 0.2" diffuse="0.8 0.8 0.8" specular="0.3 0.3 0.3" castshadow="false" pos="0 0 4" dir="0 0 -1" name="light0"></light>
	</worldbody>
    <!-- Camera settings -->
    <visual>
        <!-- allows the camera to fit into smaller spaces without clipping through -->
        <map znear="0.001" zfar="1000"/>
    </visual>
    <statistic extent="1"/>

</mujoco>

```

</details>

Here is a video, illustrating my question:
[mocap_issue.webm](https://user-images.githubusercontent.com/54349793/227456269-a16caeeb-b118-406d-bfea-79aa6a633f33.webm)

At the same time, I use the simulate function you provide to check my environment, the screenshot is as following:
[image]

Rancho-zhao, 2023-03-24:
When I change the "relpose" attribute in "weld", like this:
```
<weld body1="robot0:mocap" body2="gripper_base" relpose="0 1 0 1 0 0 0"></weld> 
``` 
the bodies act as following video:
[mocap_issue_1.webm](https://user-images.githubusercontent.com/54349793/227458244-7c03ea32-924a-4c58-9e09-ea1c312089c9.webm)

yuvaltassa, 2023-03-24:
Experiment a little with [this model](https://github.com/deepmind/mujoco/blob/main/test/engine/testdata/weld.xml) and see if it helps you. Note that you can activate the "equality" visualisation in simulate that will help you understand what's happening. I think if you carefully read the documentation of [relpose and anchor](https://mujoco.readthedocs.io/en/latest/XMLreference.html#equality-weld-relpose), you be able to figure it out.

closed 2024-01-03 by kevinzakka