google-deepmind/mujoco #791: Getting error when calling mujoco.Renderer(model) after fetching mujoco==2.3.3

rongakowang, 2023-03-24:
I am getting a error message as below:

```
Pass 'Combine redundant instructions' is not initialized.
Verify if there is a pass dependency cycle.
Required Passes:
Segmentation fault (core dumped)
```

Simplying calling the mj_step will not have any error, the error occurs only after calling the Renderer.

This error occurs after I did `pip uninstall mujoco` from my local build (from source) and fetched with `pip install mujoco==2.3.3`. But I am not sure where the error is from.

saran-t, 2023-03-24:
The only thing that I can think of that could plausibly give rise to that error is llvmpipe, which implies that you're using software (non-GPU) rendering on Linux. Does that sound right?

rongakowang, 2023-03-24:
Hi @saran-t : 

I am working on Linux, but I am not exactly sure if I am using software renderering (is there a way to confirm this?).

Also, if that is true, how may I fix this issue?

saran-t, 2023-03-24:
Try setting the environment variable `MUJOCO_GL=EGL`.

rongakowang, 2023-03-24:
Hi:

I have tried to set the variable but then come with error `'GLXPlatform' object has no attribute 'EGL'`.

saran-t, 2023-03-24:
Need a stack trace please.

rongakowang, 2023-03-24:
Sorry, the full stack trace is as:

```
Traceback (most recent call last):
  File "train/train_mujocohand_double.py", line 11, in <module>
    from anakin.artiboost import ArtiBoostLoader
  File "/home/rongakowang/MuJoCoHand/ArtiBoost/anakin/artiboost/__init__.py", line 1, in <module>
    from .artiboost_loader import ArtiBoostLoader
  File "/home/rongakowang/MuJoCoHand/ArtiBoost/anakin/artiboost/artiboost_loader.py", line 32, in <module>
    from anakin.metrics.val_metric import ValMetricAR2, ValMetricMean3DEPE2
  File "/home/rongakowang/MuJoCoHand/ArtiBoost/anakin/metrics/__init__.py", line 6, in <module>
    from .physicsmetric import DexYCBPhysicsMetric, IKMetric, DiscriminatorClasifyMetric, PointNetDiscriminatorClasifyMetric, DexYCBMeshPhysicsMetric
  File "/home/rongakowang/MuJoCoHand/ArtiBoost/anakin/metrics/physicsmetric.py", line 16, in <module>
    from model.mujoco_simulator import MuJoCoSimulator, MuJoCoSimulatorFast, MODEL_NV
  File "../model/mujoco_simulator.py", line 8, in <module>
    import mujoco
  File "/home/rongakowang/anaconda3/envs/artiboost/lib/python3.8/site-packages/mujoco/__init__.py", line 47, in <module>
    from mujoco.gl_context import *
  File "/home/rongakowang/anaconda3/envs/artiboost/lib/python3.8/site-packages/mujoco/gl_context.py", line 41, in <module>
    from mujoco.egl import GLContext as _GLContext
  File "/home/rongakowang/anaconda3/envs/artiboost/lib/python3.8/site-packages/mujoco/egl/__init__.py", line 31, in <module>
    from mujoco.egl import egl_ext as EGL
  File "/home/rongakowang/anaconda3/envs/artiboost/lib/python3.8/site-packages/mujoco/egl/egl_ext.py", line 27, in <module>
    from OpenGL import EGL
  File "/home/rongakowang/anaconda3/envs/artiboost/lib/python3.8/site-packages/OpenGL/EGL/__init__.py", line 2, in <module>
    from OpenGL.raw.EGL._types import *
  File "/home/rongakowang/anaconda3/envs/artiboost/lib/python3.8/site-packages/OpenGL/raw/EGL/_types.py", line 77, in <module>
    CALLBACK_TYPE = _p.PLATFORM.functionTypeFor( _p.PLATFORM.EGL )
AttributeError: 'GLXPlatform' object has no attribute 'EGL'
```

rongakowang, 2023-03-24:
Thank you, now the problem is solved!

closed 2023-03-24 by rongakowang