google-deepmind/mujoco #792: Collision margin changes the value of contact forces

KMCzajkowski, 2023-03-27:
Hi, first of all thank you for writing such a great piece of software and publishing the source code free of charge. I want to have a bit more control over contact forces. Therefore, I set the humanoid model so that feet contacts are always detected by adding
`<pair geom1="floor" geom2="foot1_left" margin="5"/>
    <pair geom1="floor" geom2="foot2_left" margin="5"/>
    <pair geom1="floor" geom2="foot1_right" margin="5"/>
    <pair geom1="floor" geom2="foot2_right" margin="5"/>`
to the humanoid xml + setting collision to predefined.
<br>
Now consider the following snippet
`with open('humanoid.xml', 'r') as f:
  xml = f.read()
model = mujoco.MjModel.from_xml_string(xml)
data = mujoco.MjData(model)

mujoco.mj_resetData(model, data)
mujoco.mj_forward(model, data)
data.qacc = 0  # Assert that there is no the acceleration.
data.qpos[2]-=0.001
mujoco.mj_inverse(model, data)
total_force = np.sum(np.abs(data.qfrc_inverse))
print(total_force)
print(len([model.geom(contact.geom2).name for contact in data.contact]))`

For the initial model it yields total force of 805 N, but it yields 6426349.67 N for the model with predefined contact. This looks rather peculiar. I anticipated that they would be the same, because the number of detected collisions and geometry pairs is the same in both cases.

Thank you in advance for your support.

yuvaltassa, 2023-03-27:
`margin` inflates the geoms (though this is currently not visualised), so of course the value of the contact will change. You probably want both margin and [gap](https://mujoco.readthedocs.io/en/latest/XMLreference.html#body-geom-gap)? The model described [here](https://mujoco.readthedocs.io/en/latest/XMLreference.html#actuator-adhesion) has both margin and gap, maybe play with it to understand what these two attributes do.

closed 2023-03-27 by yuvaltassa

KMCzajkowski, 2023-03-28:
Thank you for your response. I thought that it simply is the threshold for collision detection as described [here](https://mujoco.readthedocs.io/en/stable/XMLreference.html#contact-pair-margin), but it appears that it is not the case.