google-deepmind/mujoco #793: Opening URDF with STL files in same directory crashes in macOS

KyleQ1, 2023-03-30:
Hi,

I'm a student trying to use MuJoCo for a robotics college club looking to simulate legs for a person in need.

I'm looking for some help loading a URDF with other STL files included within it. I converted the Xacro to a URDF and put the STL files in the same directory. It seems to crash on startup when I try to load it.

Other questions I have: 

- Is there some way to spawn the model through terminal?

- Is this able to be used with ROS2 to be used with a launch file and have the model give joint and robot state positions?

Here is the [model](https://github.com/MaxLewter16/LLEAP/blob/main/preliminary-simulation/models/lleap_exo.urdf.xacro) and the [STL](https://github.com/MaxLewter16/LLEAP/tree/main/preliminary-simulation/meshes) files which explain my question.

[Crash Log](https://github.com/deepmind/mujoco/files/11107120/simulate_2023-03-29-182101_Kyles-MacBook-Air-10.wakeups_resource.diag.txt) converted to TXT

nimrod-gileadi, 2023-04-20:
It might be more helpful to attach the URDF rather than Xacro - we don't work with these conversion tools ourselves.

> Is there some way to spawn the model through terminal?

You can run the Simulate binary through the command line and pass a file path to the model as a command line argument. The terminal logs would be more helpful than the log you attached, I think:

```
./MuJoCo.app/Contents/MacOS/simulate /path/to/my/model
```

kevinzakka, 2023-04-30:
Closing for now, do re-open if you fail to load the STL.

closed 2023-04-30 by kevinzakka