google-deepmind/mujoco #794: Question about the qpos of a object's free joint

cidxb, 2023-03-30:
Hi, friends.

I am curious about changing a free joint qpos in an object body, shouldn't it also change the object body itself?

I thought so,  but the test result is showing that changing the free joint of an object's body cannot change the object's body position.

Is that the intended design? I thought the meaning of free joint is to control an object's body's pose.

I know I can use mujoco.data.body().xpos to change an object's body position, but I am curious about the functionality of the free joint.

Here is the test script and result :
The Body is called cup_400 and it has a free joint of the same name.
  
           ` <body name="cup_400" pos="0.59 0.55 0.88" euler="0 0 0"

            <geom mesh="cup_400_1" type="mesh" group="2"/>

            <geom mesh="cup_400_2" type="mesh" group="2"/>

             ......

            <joint armature="0" damping="0.001" name="cup_400" pos="0 0 0" stiffness="0" type="free"/>  `

Then I have define a function to randomizing the jnt qpos:
```
def randomizing_obj(self,obj_name:[str,int],quiet=False):       
 """randomizing object XY position only """
        obj_qpos=self.data.jnt(obj_name).qpos.copy()
        obj_qpos_old=obj_qpos.copy()
        obj_qpos[:2] = np.random.uniform(low=-0.2, high=0.2, size=2)  ###只改变0和1位置XY,

        self.data.jnt(obj_name).qpos[0]=self.data.jnt(obj_name).qpos[0]+obj_qpos[0]
        self.data.jnt(obj_name).qpos[1]=self.data.jnt(obj_name).qpos[0]+obj_qpos[1]

        if not quiet:
            print("\nThe random num are:",obj_qpos)
            print("\nOld qpos of jnt  :",obj_qpos_old)
            print("\nJnt qpos:",self.data.jnt(obj_name).qpos)
            print("\nThe xyz of cup is {} ".format(self.get_current_xyz("cup_400")))
```

The function `self.get_current_xyz("cup_400")`, is a function to get the XYZ position of this object now.

And the result of the function is :

```
The random num are: [ 0.12575 -0.03471  0.82991  1.       0.00001 -0.00001  0.00002]

Old qpos of jnt  : [ 0.59     0.55001  0.82991  1.       0.00001 -0.00001  0.00002]

Jnt qpos: [ 0.71575  0.68104  0.82991  1.       0.00001 -0.00001  0.00002]

The XYZ of cup is [0.59    0.55001 0.82991] 
```

As you can see, before the changing of qpos of Jnt ("cup_400"), the jnt qpos is the same as the xpos of the object, but after the changing of the qpos of the jnt, the position of the body is not changed .

Is joint position and body position supposed to act like this?

Thank you for your answer! I am just very curious.

nimrod-gileadi, 2023-03-30:
Can you paste the implementation of `get_current_xyz`?

Generally, after changing qpos, one needs to run forward-kinematics to update the cartesian positions of bodies, using `mj_kinematics` (if you just need positions and velocities) or `mj_forward` (if you want accelerations).

closed 2023-03-30 by nimrod-gileadi

cidxb, 2023-03-31:
> Can you paste the implementation of `get_current_xyz`?
> 
> Generally, after changing qpos, one needs to run forward-kinematics to update the cartesian positions of bodies, using `mj_kinematics` (if you just need positions and velocities) or `mj_forward` (if you want accelerations).

Of course, actually, I just wrap the `self.data.body(body_name).xpos` inside the `get_current_xyz`.

And you are absolutely right!
After I added` mujoco.mj_forward(self.model,self.data)`, the XYZ of the object does change according to the free joint.

So I get it, the data parameter xpos is the direct parameter of the cartesian positions of bodies, so when it changes it is effective immediately, but joint qpos need the forward-kinematics to quasi to calculate the cartesian positions.

Thank you again for the straight and clean answer!