google-deepmind/mujoco #795: How to render offscreen and window at the same time

joonhyung-lee, 2023-03-30:
Hi,

I'm a graduate student and I'm trying to use MuJoCo for offscreen view rendering.

I'm looking for some help with offscreen rendering.

As I said in the title, I want to renderer both **the window and offscreen** and get images corresponding to various views through offscreen at the same time.
However, **when I load an image with offscreen, the window render turns black, and nothing appears.**
(I'm using python for programming.)

Here is a screenshot / video, illustrating my question:

Any help would be greatly appreciated.

My simple code is below. (that How do I render offscreen

`from PIL import Image

# _mode_list = ['mujoco','mujoco_viewer','offscreen']
mode = 'offscreen'

if mode == 'offscreen':
    env.viewer = mujoco_viewer.MujocoViewer(env.model, env.data,
                                                'offscreen',
                                                width = env.viewer.viewport.width, height = env.viewer.viewport.height)
    rendered_frames = []

if mode == 'mujoco_viewer':
    env.viewer = mujoco_viewer.MujocoViewer(env.model, env.data,
                                                width=env.viewer.viewport.width, height=env.viewer.viewport.height)
    
if mode =='offscreen':
    env.viewer.cam.azimuth = 0
    env.viewer.cam.distance = 3.5
    env.viewer.cam.elevation = -70
    env.viewer.cam.lookat[0] = 0.0
    env.viewer.cam.lookat[1] = 0.0
    env.viewer.cam.lookat[2] = 1.2

    frame = env.viewer.read_pixels()
    frame = Image.fromarray(frame)
    
    rendered_frames.append(frame.copy())

if mode == 'mujoco_viewer':
    env.viewer.render()

if mode == 'mujoco':
    pass`

closed 2023-04-25 by saran-t

fukangl, 2023-10-23:
@saran-t Can you provide instructions on how to save images from the viewer or a camera?

yuvaltassa, 2023-10-23:
The `simulate` app has a "Screenshot" button that saves a screenshot to PNG file, is that what you're looking for? I'm not sure if it works with the passive viewer, I've never tried that.

Or do you need it programatically? If the latter then you can just do an offscreen render from Python and save that to file using Python tools.

joonhyung-lee, 2023-10-24:
@fukangl Here's how I did it! I used the `mujoco.mjr_readPixels()` function to render on-screen.
[demo_ur_get_image.zip](https://github.com/google-deepmind/mujoco/files/13110137/demo_ur_get_image.zip)

fukangl, 2023-10-24:
Thanks a lot! I used Renderer to save images.

louisdehaybe, 2024-12-04:
> The `simulate` app has a "Screenshot" button that saves a screenshot to PNG file, is that what you're looking for? I'm not sure if it works with the passive viewer, I've never tried that.
>
> Or do you [...]

Hello, I am trying to save images programatically. I manage to save the image of the scene, however all the geometries that have been added with python (i.e. not in the XML initially) do not appear in the image. Is there an efficient way of "copying" the viewer usr_scn to the Renderer scene in order to save an image that exactly corresponds to what I see in the viewer ? Also, are there ways to change the quality of the saved image ? 

Thanks